mobizt/Firebase-ESP32 #83: Support for Firebase Cloud Storage

TMaxedCode, 2020-07-09:
**Is your feature request related to a problem? Please describe.**
I need to upload many images taken from Esp-32 Cam (Ai-Thinker board) to Firebase Cloud Storage.

**Describe the solution you'd like**
Support for uploading images or files to Firebase Cloud Storage.

**Describe alternatives you've considered**
Using Firebase Realtime Database wouldn't be a good solution since I will be downloading many images. Using Firebase Cloud Storage, I can download 1gb of data per day for free but if I use Firebase Realtime Database, I can only download 10gb of data per month. This also includes the sensor data I regularly upload to Firebase Realtime Database.

mobizt, 2020-07-09:
I currently develop the new update which include Firebase Cloud Storage (upload/download) and more features except Cloud Firestore via gRPC which currently undeveloped.

Unfortunately I found the memory leak and trying to fix.

I will inform you when I update.

TMaxedCode, 2020-07-09:
Thank you @mobizt. I can wait for the new update. Really excited. Have a nice day :)

TMaxedCode, 2020-07-10:
@mobizt I noticed a typo error in this repo and firebase esp8266. Instead of getFloat, setFloat, and pushFloat, you used getFlot, setFlot, pushFlot.

closed 2020-07-11 by mobizt

mobizt, 2021-01-13:
Finally, Firebase Cloud storage are supported in this new library 
https://github.com/mobizt/Firebase-ESP-Client