mobizt/Firebase-ESP32 #85: HELP

Rebehy, 2020-07-21:
Hi all,

ESP32 - Firebase RTDB.

I am hoping this is something simple. I am having some trouble with readtimeout. I am able to read strings up to around 12kb but as soon as my string gets a little longer and takes over say 10 sec, it fails with a read Timeout error. I have tried to set this using;

Firebase.setReadTimeout(firebaseData, 60000);

But this still doesn't work.

Also sometimes as my string get towards the point where it stops reading I occasionally get a data type mismatch without  changing anything but the length of the string. I am fairly new to all this and any help would be appreciated.

My goal are strings of around <20kb, which I am currently transferring in packets of 20b from my Thunkable app to my ESP32 using BLE. This is taking around 3 minutes. Because I am able to write to firebase from Thunkable, reading into my ESP32 from the db would be a much preferred solution for me.

Thanks in advance.
Eddie

--
void FirebaseComm(){
  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
  Firebase.setReadTimeout(firebaseData, 60000);
  Firebase.setwriteSizeLimit(firebaseData, "large");
  int t=millis();
  if (Firebase.get(firebaseData,"/esptest")){
    Serial.println("found path");
    if (firebaseData.dataType()=="string"){
      Serial.println("getting string");
      Serial.println(firebaseData.stringData());
      Serial.println(millis()-t);
    }
    else{
      Serial.println(firebaseData.errorReason());
      Serial.println("should print error");
    }
  }
  else {
    Serial.println(firebaseData.errorReason());
    Serial.println("Can't resolve firebase");
  }
}

on success:
found path
getting string
"my long string here"
7847

on fail:
read Timeout
Can't resolve firebase

Also getting;
found path
getting string
"my long string here..."
7982
data type mismatch

mobizt, 2020-07-21:
The Firebase.setwriteSizeLimit is for database writing to server and does not related to the database reading.  
I also correct the description of function due to copy and paste error.

Please try to change the tcp client timeout defined in [FirebaseESP32HTTPClient.h on line No. 96](https://github.com/mobizt/Firebase-ESP32/blob/master/src/FirebaseESP32HTTPClient.h#L96)

From 
**uint16_t tcpTimeout = HTTPCLIENT_DEFAULT_TCP_TIMEOUT;** // default is 5000 milliseconds
to
**uint16_t tcpTimeout = 12000; //12 seconds**

Rebehy, 2020-07-21:
> Please try to change the tcp client timeout defined in [FirebaseESP32HTTPClient.h on line No. 96](https://github.com/mobizt/Firebase-ESP32/blob/master/src/FirebaseESP32HTTPClient.h#L96)
>
> From [...]

Worked!! :)
Thank you.

closed 2020-07-21 by Rebehy