mobizt/Firebase-ESP32 #86: HELP: Storing sensor weight data with 3 decimal points double

arneetsinghkalra, 2020-07-22:
I am getting input data from an HX-711 load sensor. I am pushing data to the Firebase Realtime Database as a json object containing a double of the load sensor, alongside the time and another distance sensor (int data). How do I only push 3 decimal places to the Firebase Database? Thanks!

mobizt, 2020-07-23:
You should use float instead of double.

The library does not provide the floating point number resolution settings.

If you want to do so, you can change the number of decimal places in [FirebaseESP32.cpp at line number 5022](https://github.com/mobizt/Firebase-ESP32/blob/master/src/FirebaseESP32.cpp#L5022) 

from 

**dtostrf(value, 7, 6, buf);**

to

**dtostrf(value, 7, 3, buf);** //if 3 decimal places required

****Don't modify this for double** because you will affected to all many functions and  may cause the problems because we use the resolution to identify the float and double values.

closed 2020-07-23 by mobizt

arneetsinghkalra, 2020-07-23:
I would use floats but I keep getting this error whenever I try putting a float inside a JSON object to be sent to the database: "undefined reference to `bool FirebaseJson::set<float>(String const&, float)'
sketch/ESP32_Logic_W_Intervals.ino.cpp.o: In function `sendToFirebase(String)':"

mobizt, 2020-07-23:
You can update the library which now float was supported for JSON.

arneetsinghkalra, 2020-07-23:
I tried that. When I update line 5022, it only updates the decimal points for data that is stored only as a float in the database. Any floats stored inside a JSON object are still around 7 decimal places. Any work around for that?

mobizt, 2020-07-23:
Do the same  in [this file](https://github.com/mobizt/Firebase-ESP32/blob/master/src/FirebaseJson.cpp#L322).

arneetsinghkalra, 2020-07-23:
I have tried that. It's weird, it's still not working. 
[image]

mobizt, 2020-07-23:
I can't see anything wrong and nothing behind.

I also test it.

Make sure you edit the right line of code.

mobizt, 2020-07-23:
Make sure you pass the float variable to the JSON object instead of number to ensure that the compiler will treat that value as float instead of double.

agungaufa, 2020-07-24:
> I am getting input data from an HX-711 load sensor. I am pushing data to the Firebase Realtime Database as a json object containing a double of the load sensor, alongside the time and another distance sensor (int data). How do I only push 3 decimal places to the Firebase Database? Thanks!

sorry,how about this
float foo = 12.34567;
int bar = (int) foo * 1000.0; // bar now = 12345
foo = (float) bar / 1000.0;   // foo now = 12.345

mobizt, 2020-07-24:
I don't get that.
What is the problem now?

mobizt, 2020-07-24:
I see that you push the JSON of Distance, Time and Weight children, then I recommend you to use float variable when you set or add float value to FirebaseJson object.

Example:
```C++

FirebaseJson json;
float myWeight = 123.456789

json.add("Weight",myWeight);

Firebase.push(firebaseData, "Path", json);
```