mobizt/Firebase-ESP32 #87: Arduino\libraries\Firebase_Arduino_based_on_WiFiNINA\src/Firebase_Arduino_WiFiNINA.h:51:26: fatal error: avr/pgmspace.h: No such file or directoryBUG

kenken64, 2020-07-28:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.9]

**ESP8266 Arduino Core SDK version**
 - Version [e.g. 2.5.1]

**Additional context**
Add any other context about the problem here.

mobizt, 2020-07-28:
You are using the unsupported or incompatible device with this library.

Please read the readme and check your device (board) that supported by the library.

mobizt, 2020-07-28:
This is for ESP32 platform. The error you mentioned is for AVR/ARM platforms.

closed 2020-07-28 by mobizt