mobizt/Firebase-ESP32 #88: HELP: How to set datetime on json

kenken64, 2020-07-29:
FirebaseJson how to set ?

mobizt, 2020-07-29:
No datetime data type supported in Firebase RTDB.

It supports primitive types e.g. boolean, number (int, float, double), string, json object and array

You can store it as string or integer of UNIX timestamp.

If you are new to Firebase RTDB please read the google document, open the Firebase console (web) and try with it and see how is it working.

mobizt, 2020-07-29:
There are examples which you can try [here](https://github.com/mobizt/Firebase-ESP32/tree/master/examples).

Read the readme doc how to use the library or check the available functions [here](https://github.com/mobizt/Firebase-ESP8266/tree/master/src).

closed 2020-07-29 by mobizt