mobizt/Firebase-ESP32 #90: HELP-Timestamp example not working

noonmasho, 2020-08-04:
Hi,  I was using this example file-https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Timestamp/Timestamp.ino

But it is showing an error as given below. I'm able to send Json object succesfully using your library.But for the part where connection has to be established it is showing this error. Please guide me. Thank you in advance and thank you for your library.

------------------------------------
Set Timestamp test...
FAILED
REASON: connection refused
------------------------------------

------------------------------------
Get Timestamp (double of milliseconds) test...
FAILED
REASON: connection refused
------------------------------------

------------------------------------
Push Timestamp test...
FAILED
REASON: connection refused
------------------------------------

mobizt, 2020-08-04:
I did not found any error for that example.

The connection refused is the result from invalid database secret and/or invalid Firebase host.

Please post your working code here.

noonmasho, 2020-08-04:
Hi, This particular snippet worked in another code I tried
```

int W = 32;
 int L = 56;
  
FirebaseJson json;

json.set("W",W);
json.set("L", );
Firebase.pushJSON(firebaseData, "/Test_1", json);
```

The other code I tried for which I faced the problem is exactly the same as given here https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Timestamp/Timestamp.ino

I made changes only to the wifi credentials and firebase host and AUTH. 

The read and write rules are true

Thank you in advance

mobizt, 2020-08-04:
No error for that example which I have been tested it.

closed 2020-08-04 by mobizt

noonmasho, 2020-08-04:
Hi, I understand that there is no problem with the example. I request you to help me to understand the issue here. I can't understand the reason for error connection refused. Please do help. If Firebase.pushJSON() is working why is it not working for this?
Thank you

mobizt, 2020-08-04:
I don't know. The error is the server refuse to connect because the authentication error as I said above.

You should post your code even it is the example code and replace the database secret with something before post.

noonmasho, 2020-08-04:
Hi, this is the code
```
#include <WiFi.h>
#include <FirebaseESP32.h>

#define WIFI_SSID "xxxx"
#define WIFI_PASSWORD "yyyyyy"  
#define FIREBASE_HOST "xxxxxx"
#define FIREBASE_AUTH "yyyyy"

//Define Firebase Data object
FirebaseData firebaseData;

void setup()
{

  Serial.begin(115200);
  Serial.println();
  Serial.println();

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  /*
  This option allows get and delete functions (PUT and DELETE HTTP requests) works for device connected behind the
  Firewall that allows only GET and POST requests.
  
  Firebase.enableClassicRequest(firebaseData, true);
  */

  String path = "/Timestampcheck";

  Serial.println("------------------------------------");
  Serial.println("Set Timestamp test...");

  if (Firebase.setTimestamp(firebaseData, path + "/Set/Timestamp"))
  {
    Serial.println("PASSED");
    Serial.println("PATH: " + firebaseData.dataPath());
    Serial.println("TYPE: " + firebaseData.dataType());

    //Timestamp saved in millisecond, get its seconds from intData()
    Serial.print("TIMESTAMP (Seconds): ");
    Serial.println(firebaseData.intData());

    //Or print the total milliseconds from doubleData()
    //Due to bugs in Serial.print in Arduino library, use printf to print double instead.
    printf("TIMESTAMP (milliSeconds): %.0lf\n", firebaseData.doubleData());

    //Or print it from payload directly
    Serial.print("TIMESTAMP (milliSeconds): ");
    Serial.println(firebaseData.payload());

    //Due to some internal server error, ETag cannot get from setTimestamp
    //Try to get ETag manually
    Serial.println("ETag: " + Firebase.getETag(firebaseData, path + "/Set/Timestamp"));
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  Serial.println("------------------------------------");
  Serial.println("Get Timestamp (double of milliseconds) test...");

  if (Firebase.getDouble(firebaseData, path + "/Set/Timestamp"))
  {
    Serial.println("PASSED");
    Serial.println("PATH: " + firebaseData.dataPath());
    Serial.println("TYPE: " + firebaseData.dataType());

    printf("TIMESTAMP: %.0lf\n", firebaseData.doubleData());
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  Serial.println("------------------------------------");
  Serial.println("Push Timestamp test...");

  if (Firebase.pushTimestamp(firebaseData, path + "/Push/Timestamp"))
  {
    Serial.println("PASSED");
    Serial.println("PATH: " + firebaseData.dataPath());
    Serial.print("PUSH NAME: ");
    Serial.println(firebaseData.pushName());

    //Due to some internal server error, ETag cannot get from pushTimestamp
    //Try to get ETag manually
    Serial.println("ETag: " + Firebase.getETag(firebaseData, path + "/Push/Timestamp/" + firebaseData.pushName()));
    Serial.println("------------------------------------");
    Serial.println();
  }
  else
  {
    Serial.println("FAILED");
    Serial.println("REASON: " + firebaseData.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
}

void loop()
{
}
```

mobizt, 2020-08-04:
Are you sure the Firebase Host is valid and no slash (/) at the end?

noonmasho, 2020-08-04:
Hi, Thank you so much . That was the issue.  Thank you for your time. You are a great person. Thank you so much.