mobizt/Firebase-ESP32 #92: ESP32 reset with wifi connection lost and firebase

am-canes, 2020-08-10:
I am having a problem with ESP32 when firebase is processing the result of a stream or when it is doing a setTimestamp and wifi connection is lost. 
The board resets with the error:

[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400eb231  PS      : 0x00060c30  A0      : 0x800eb361  A1      : 0x3fff2bb0  
A2      : 0x3ffc1ebc  A3      : 0x3ffc1db8  A4      : 0x3ffc1fb8  A5      : 0x3ffc2c40  
A6      : 0x3ffc3158  A7      : 0x3ffc24b8  A8      : 0x800eb231  A9      : 0x3fff2b90  
A10     : 0x00000000  A11     : 0x3ffc1db8  A12     : 0x3f424a2b  A13     : 0x3ffb1e90  
A14     : 0x00000000  A15     : 0x3fff07a0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400eb231:0x3fff2bb0 0x400eb35e:0x3fff2c50 0x400eb7ee:0x3fff2c70 0x400eb80f:0x3fff2e00 0x40089e81:0x3fff2e20

Sometimes the error [E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C) occurs and the board doesn' reset, it is like just a warning.

When my system is installed in a place with bad connection, the board resets many times.

Anyone could help me to solve this problem?
Thank you

mobizt, 2020-08-10:
The error **Exception was unhandled** is the coding error.

It's hard to say what are the causes of problems, it may relate to improper use of object or variable or trying to access the variable or object that was not initialize or does not exist anymore after initiated which may relate or not relate to your library usages.

Please post your code to let me help to identify the root causes.

am-canes, 2020-08-11:
@mobizt my code is big so I will try to explain the main of it.
I'm using two cores. At core 0 I'm monitoring my sensors, communicating with an IO expander and updating a display through I2C, using hardware timer and saving some parameters at EEPROM when it changes.
At core 1 I'm starting and handling the wifi and firebase comunication. 

The problem is with something that I'm doing inside firebase code because if I comment it the code works well with wifi connection lost.
So I will show my firebase code. I am using two instances of FirebaseData, one for stream and other for set data. I have a FirebaseJson global to do a backup in case of error and the others json variables are local.
I changed printResult on the stream callback to process what I need, but the error occurs with this function commented so it is not the problem.
Function processFirebase is called on core 1 main function and each 1 second, if there is any change, it updates data using updateNodeSilent and setTimestamp in sequence.

`FirebaseData firebaseData1;
FirebaseData firebaseData2;

FirebaseJson jsonbkp;

void streamCallback(StreamData data)
{
    Serial.println("Stream Data1 available...");
    printResult(data);
}

void streamTimeoutCallback(bool timeout)
{
    if (timeout)
    {
        Serial.println("Stream timeout, resume streaming...");
    }
}

void initFirebase() 
{
    Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
    firebaseConnected = true; 
    Firebase.reconnectWiFi(true);
    path = "/" + devMacId + "/config";
    Serial.print("path: ");
    Serial.println(path);

    initData();

    if (!Firebase.beginStream(firebaseData1, path))
    {   
        Serial.println("------------------------------------");
        Serial.println("Can't begin stream connection...");
        Serial.println("REASON: " + firebaseData1.errorReason());
        Serial.println("------------------------------------");
        Serial.println();
    }

    Firebase.setStreamCallback(firebaseData1, streamCallback, streamTimeoutCallback);    
}

void initData()
{
  FirebaseJson json;
  
  json.set("config/time_fan", (unsigned short)tempo_fan);
  json.set("config/time_pump", (unsigned short)tempo_bomba);
  json.set("config/time_flame", (double)tempo_chama);
  json.set("config/attempts", (unsigned short)attempts);
  json.set("config/setpoint", (double)setpoint);
  json.set("temperature", (double)temperature);

  if (Firebase.setJSON(firebaseData2, "/"+devMacId, json))
  {
      Serial.println("Iniciando o envio de dados...");
  }
  else
  {
      Serial.println("Falha no envio inicial de dados");
      Serial.println("REASON: " + firebaseData2.errorReason());
  }
}

void processFirebase() 
{
  FirebaseJson jsonUpdt;
  if (millis() - sendDataPrevMillis > 1000)
  {
    sendDataPrevMillis = millis();
    if(errorToSend)
      jsonUpdt = jsonbkp;

    if(firebaseStruct.updateSetpoint)
    {
        firebaseStruct.updateSetpoint = false;
        jsonUpdt.add("config/setpoint", setpoint);
        updateJson = true;
    }
    if(firebaseStruct.updateTemp)
    {
        firebaseStruct.updateTemp = false;
        jsonUpdt.add("temperature", tempCloud);
        contTempCloud = 0;
        updateJson = true;
    }
    if(firebaseStruct.updateFlame)
    {
        firebaseStruct.updateFlame = false;
        jsonUpdt.add("config/time_flame", tempo_chama);
        updateJson = true;
    }

    if(updateJson)
    {
      updateJson = false;
      if (Firebase.updateNodeSilent(firebaseData2, "/"+devMacId, jsonUpdt)) //decidir se utiliza setJSON ou updateNode
      {
          secondsFirebase = 0;
          Serial.println("Update node");
          if(errorToSend)
          {
            errorToSend = false;
            jsonbkp.clear();
            Serial.println("send error");
          }
      }
      else
      {
          Serial.println("Falha - Update node");
          Serial.println("REASON: " + firebaseData2.errorReason());
          errorToSend = true;
          jsonbkp = jsonUpdt;
      }

      if (Firebase.setTimestamp(firebaseData2, "/" + devMacId + "/timestamp"))
      {
        Serial.println("Set timestamp");
      }
      else
      {
        Serial.println("Falha - Set timestamp");
        Serial.println("REASON: " + firebaseData2.errorReason());
      }
    }

    if(secondsFirebase >= 3600  || sendStatus) 
    {
        if(sendStatus)
        {
          jsonUpdt.add("config/getStatus", false);
          Serial.println("send status");
        }
        else
          Serial.println("keep alive");

        secondsFirebase = 0;
        jsonUpdt.add("temperature", temperature);
        jsonUpdt.add("status_system", flag_On); 
        jsonUpdt.add("status_heater", flag_status);   
        if (Firebase.updateNodeSilent(firebaseData2, "/"+devMacId, jsonUpdt))
        {
            tempCloud = temperature;
            contTempCloud = 0;
            secondsFirebase = 0;
            sendStatus = false;
        }
        else
        {
            Serial.println("Falha - keep alive / send status");
            Serial.println("REASON: " + firebaseData2.errorReason());
        }

        if (Firebase.setTimestamp(firebaseData2, "/" + devMacId + "/timestamp"))
        {
           Serial.println("Set timestamp");
        }
        else
        {
          Serial.println("Falha - Set timestamp");
          Serial.println("REASON: " + firebaseData2.errorReason());
        }
    }
  }
}`

Thank you for your assistance. I hope that you could help me.

mobizt, 2020-08-11:
As I said, you need to check and take care of  your free heap.

mobizt, 2020-08-11:
Print free heap to serial before and after calling the Firebase function.

If you can't provide  the full code that causes the problem, you may need to separate the code for the Firebase from the other part of your code.

Simulate the fake sensor data to use for the Firebase and run only Firebase related code which you can test it yourself or share it to me.

am-canes, 2020-08-11:
Ok, I will run some tests and share the results and the code with you tomorrow.

am-canes, 2020-08-12:
@mobizt I did some testes and the error is happening even with a simple code. 
I shared the code here: https://github.com/am-canes/testFirebase.
See if you can download and compile it. I'm using platformIO and board Espressif ESP32 Dev Module.
You just need to config your wifi and firebase connection.

In the code I'm updating a value each 10 seconds and I have a stream too. 
To test it easily I'm routing my phone internet and I stop the connection when the code is receiving data from stream or at setTimestamp. Sometimes it resets when the connection is lost and sometimes when the wifi connects again.
When you see on serial "Update node" you can drop connection. Another test is when you see on serial that there are data to receive on the stream.

Im my original code, the free heap is around 115000 to 155000.
I'm printing free heap in this other code too.

Thanks.

mobizt, 2020-08-13:
Your free heap is the problem as SSL library required about 100k for operation.

am-canes, 2020-08-13:
Is there anything that I can do to solve this?
What could be using so much heap even in my simple code?

mobizt, 2020-08-13:
The FirebaseJson.h does not need to include because it includes already in the library.

Your  Firebase initialize function **initFirebase** and WiFi setup should be in setup (main task) function instead of in rtos task.

In **processFirebase** function, you may read and/or averaged your sensor data frequently as you want but sending data to server every 1 second is not recommend.

Use  **vTaskDelay** in coreTaskOne too.

am-canes, 2020-08-13:
I did the corrections that you suggested and the free heap is still around 115k, mainly after the stream callback.
Sometimes after set some value it is around 155k.
But even in the simple code the free heap is around 135k with wifi and firebase and the two tasks.
Could it be related with the use of the two cores? Any configuration of the tasks?

What is the interval recommended to send data to server?

mobizt, 2020-08-13:
Let the Firebase library runs on main task is recommend and the library also use the internal rtos task to process the stream.

> What is the interval recommended to send data to server?

This is the remote server communication not a local server, you need to think of the bandwidth of your usage and the necessary for do that which depends on you but for me it's too much and nonsense.

mobizt, 2020-08-13:
Follow the stream callback example and check the free heap.

mobizt, 2020-08-13:
Try to reduce the preserved stack size (in bytes) for rtos task for coreTaskZero.

Delete the unnecessary coreTaskOne task and move the processFirebase to loop function of main task.

Note:
The almost memory used by the mbedTLS and tcp client which initiated in the FirebaseData object.

am-canes, 2020-08-15:
Sorry for the delay, I was running the tests.
I did what you suggested.
I reduced the stack for coreTaskZero and moved firebase functions to loop function deleting coreTaskOne.
Then I tested it with the stream callback example. The free heap is between 150k and 200k but sometimes, depending when the wifi conection is lost, the board resets even with a free heap of 195k. That's not enough heap for SSL library?

mobizt, 2020-08-16:
If you're using the latest ESP32 core SDK and the latest library version.

The board voltage may fluctuate or drop during WiFi operation due to the high internal impedance of power supply.

If you're using the USB cable to supply your board, please try to replace with shorter cable.

Use the large value capacitor e.g. 220 uF or more connected between 3.3V and GND on your board.

am-canes, 2020-08-16:
I think that I'm using the latest version:
PLATFORM: Espressif 32 1.12.2 > Espressif ESP32 Dev Module
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)

WiFi 1.0 and WiFiClientSecure 1.0.

I replaced the USB cable and tried with a capacitor of 220uF and 470uF but it keeps happening sometimes. 
I will test tomorrow in my real system with a power supply instead of USB cable to see how it works. Maybe my test system is causing some problem that I don't have in my real system and the modifications can work.

I appreciate if do you have any other ideas that I can test.

mobizt, 2020-08-17:
The software issue is rarely the cause which the library is regularly updated and has been tested. 

The voltage ripple or noisy or weak power supply is the most possible causes.

am-canes, 2020-08-20:
I tried the changes in my real system with a good power supply and it still happens. 
I noted that it occurs when I'm using the stream and the wifi connection is lost when nothing is being processed, when it is waiting for another stream. I'm trying to find exactly where the error happens but I still coudn't find anything. Could you guide me where to look on Firebase code?

Another thing that I noted is that when the wifi reconnects and receive the fisrt stream data, it is interfering in core zero. Because I have a sensor that is activated by a pulse through I2C and I can see it has a little delay. Do you know what could be causing this? 

About the free heap, there is any function that I can clean the garbage from the heap when I see it is low?

mobizt, 2020-08-20:
I has been tested frequently in many devices as I stated in the README and under many conditions and I did not found the error.

Nothing in the library related to your concerns as the internal stream task [attaches to core 1](https://github.com/mobizt/Firebase-ESP32/blob/master/src/FirebaseESP32.cpp#L4830).

The most memory used in Firebase Data object is mbedTLS and the two of internal string buffer of user data and the json object.

The internal string buffer can be clear with **firebaseData.clear** but the cached data will be gone.

mobizt, 2020-08-20:
Make sure you understand what I mention about supply voltage correctly.

The constant voltage needed for mcu during high current draw by WIFI operation which the strong power supply can't help if you still use long and high impedance USB cable then I recommend to use shorter one or make your board supply voltage more stable by decrease it's internal impedance by use large capacitor connected across 3.3V and GND.

am-canes, 2020-08-20:
Yes, in my protoboard I'm using a USB cable, I changed by one shorter and used a 470uF capacitor between 3.3V and GND but nothing changed.

In my real system I'm using a power supply of 12V with a regulator to 5V for the ESP32 on Vin pin. I used the 470uF capacitor between 3.3V and GND too. But the reset keeps happening sometimes when if wifi connection is lost. 

I thought it could be something related with the other things that I am using in my code, but it still happens with the simple code just with wifi connection and firebase stream callback example.

I'm using ESP32 DEVKIT V1, can it be something realated to the kit?

mobizt, 2020-08-20:
This never happened to me and should not happen.

If you're using PlatformIO, please remove other installed libraries and try again.
Or use Arduino IDE or try to change the device if possible.

am-canes, 2020-09-06:
I have been testing with other devices, I've tried with 5 different devices and the problem still occurs in all of them. 

Any other idea of what can I test?

mobizt, 2020-09-06:
No everything I recommended already and you should find the root cause yourself.  It's upon your environment and the peripheral devices that use with your devices e.g. strong high frequency magnetic field interference, noisy and weakness supply bus voltage (I already note).

mobizt, 2020-09-13:
After deep investigation for a week with this issue in platformIO, I 've found that the causes are from the WiFiClientSecure and mbedTLS.

The issue fixed and please try the update.

am-canes, 2020-09-14:
Good news! Thank you for the correction.

I updated the library today and tested it all day. It seems more stable but the reset happened tree times during the day. But today I was testing the board with USB supply. Tomorrow i will test it in my system with a good power supply.
I will let you know about the results.

Thanks.

mobizt, 2020-09-14:
The exception still be happened in the core library side of mbedTLS during the SSL/TLS handshake and WiFi drops connection during that process.

am-canes, 2020-09-16:
Okay, I understood. I will open an issue on the github of the core to see if they can help me.

Do you said that you find the problem in platformIO. Do you have any other core that you use on Arduino IDE that works without problem?
Thanks.

mobizt, 2020-10-04:
If you still interested in this issue, I create a new [test-branch](https://github.com/mobizt/Firebase-ESP32/tree/test-branch) to test the WiFiClientSecure and mbedTLS related issue there.

When compiled with PlatformIO, it will print all the debug information to serial.

This branch will defer the WiFi reconnection for 10 seconds after the WiFi was disconnected and available. 

The modified version of WiFiClientSecure and ssl_client core libraries are included.

If you still get the exception error like this

```
Debug exception reason: Stack canary watchpoint triggered (stream_task0)
```

This depends on the payload handle in the stream task which the stack to hold the stream data when the system interrupted may overflow.

To increase the stack size for stream task, in the sketch use this function in the setup().

```C++
Firebase.setStreamTaskStackSize(20000); //20000 bytes of stack memory reserved 
```

mobizt, 2020-10-04:
From long testing, this issue also happen when create the Firebase stream connection using the WiFiClientSecure class without  using the Firebase library.

The custom test code was just simple and clean, the seldom issue still occur when WiFi disconnected.

The test-branch repo uses the modified WiFiClientSecure for proper releasing of the mbedTLS used resources when connection was closed or disconnected.

am-canes, 2020-10-05:
Yes, I am still interested in this issue. 
I will run some tests with the new branch this week.
Thank you

mobizt, 2020-10-05:
The multitasking and memory handle for incoming socket connection are come into mind. The parallel tasks of the consequence connections need to be queued or blocked for operation.  The accepted connection response handle should be queued to process later. At the current implementation of the library, it performs all accepted connections simultaneously which may cause the memory issue.

closed 2020-10-06 by mobizt

mobizt, 2020-10-06:
The library was updated on the master branch which now the unhandled exception was fixed already (believe that).

The issue is about the reconnect feature when the WiFi disconnected was detected which WiFi.reconnect() was called internally and can cause this issue if it called immediately after the socket closed the connection and mbedTLS resources are released. 

This issue is hardly to detect and seldom occurring because of the reconnect will be called when other new request take places which can be called from the stream task request loop in the multitasking support mcu.

The WiFi.reconnect() function will disconnect and connect the WiFi using the saved WiFi credentials. Don't know exactly why the calling WiFi.reconnect() immediately after the socket connection closed can cause the unhandled exception error.

The fixes are the delay of calling WiFi.reconnect() when the socket connection closed and preventing the release of the uninitialized resource in the WiFiClient library which also leads to the issue.

You can update the library v3.8.1 and test it.

am-canes, 2020-10-07:
I updated the library to version 3.8.1 and I'm testing since yesterday.
I didn't have any reset caused by the wifi connection lost.

Only one strange behavior happen sometimes, I'm having:
`
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (task_stream0)
`
Even when I tried to change setStreamTaskStackSize to 50000.
Now after some tests it has stopped. I will test more and I return to give a feedback.
I'm doing a long test with other boards too.

Thank you for the efforts to solve the problem!

mobizt, 2020-10-08:
Stack size issue (overflown), you can increase here
**Firebase.setStreamTaskStackSize(20000); //20000 bytes of stack memory reserved** 

Unfortunately I test it frequently and all the time (switching on and off WiFi hotspot) and the exception still be existed but very rarely and can't control in the library side.

am-canes, 2020-10-19:
Yes, I increased but sometimes it overflown anyway.
The problem is that when the wifi connection is bad it is not so rarely and happen a lot of times. But I understand that you can't control on library side.