mobizt/Firebase-ESP32 #93: HELP

mars000, 2020-08-16:
thanks for pulling together a great library.
Wonder - why is it (at startup especially) it can take 2-3 times longer to do a setString vs setInteger ? what is going on under the hood.

Also - any way to make setString, setInt non blocking ?

kind regards

mobizt, 2020-08-16:
It's not different in term of operation between that functions. 

The time consumed for operation based on the network situation.

closed 2020-08-16 by mobizt

mobizt, 2020-08-16:
The client should take care and wait for the server response to be finished with the timeout to ensure the success of the request.

The time consuming then vary depending on the time used in request , response and SSL/TLS handshake.

If you are using different Firebase Data object for streaming and reading/storing data, the connection between server and client is kept alive and the time consuming for the request and response will be significantly reduced (approx. 200 - 300 ms per request).

mars000, 2020-08-16:
thank you mobizt for the quick and helpful response.

A few questions:

1. Is there a non blocking way of writing to firebase ?
2. How long does the websocket stay open (I assume its a keepalive and stays open for as long as the app is running and internet connected via wifi) ?
2. I've edited slightly your Blynk example. Given your above advice I'm using different Firebase data objects for each Stream and write - is this best practice for making it fast as possible ?

My code below (modified from your ;-))

```

/*
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt
 * 
 * Copyright (c) 2020 mobizt
 * 
 * This example is for the beginner
 *
*/

#include <WiFi.h>
#include <FirebaseESP32.h>
#include <Ticker.h>

//1. Change the following info
#define FIREBASE_HOST "https://esp32-flutter-firebase-ec9c6.firebaseio.com/"
#define FIREBASE_AUTH "xxxxx"
#define WIFI_SSID "xxx"
#define WIFI_PASSWORD "xxx"

Ticker sendMessage1;
Ticker sendMessage2;

//Debug Blynk to serial port
#define BLYNK_PRINT Serial

#include <WiFiClient.h>
#include <BlynkSimpleEsp32.h>

//Auth token for your Blynk app project
#define BLYNK_AUTH "YOUR_BLYNK_APP_PROJECT_AUTH_TOKEN"

//Define FirebaseESP32 data objects
FirebaseData firebaseData0;
FirebaseData firebaseData1;
FirebaseData firebaseData2;
FirebaseData firebaseData3;

String path0 = "/Blynk_Test";
String path = "/Blynk_Test/Int";
String path1 = "/Blynk_Test/String";

//D4 or GPIO2 on Wemos D1 mini
uint8_t BuiltIn_LED = 2;

void printResult(FirebaseData &data);
void printResult(StreamData &data);

/*
Blynk app Widget setup 
**********************
1. Button Widget (Switch type), Output -> Virtual pin V1
2. LED Widget, Input -> Virtual pin V2
*/

[... 313 lines cut ...]
    Serial.println();

    for (size_t i = 0; i < arr->size(); i++)
    {
      Serial.print(i);
      Serial.print(", Value: ");

      FirebaseJsonData *jsonData = data.jsonDataPtr();
      //Get the result data from FirebaseJsonArray object
      arr->get(*jsonData, i);
      if (jsonData->typeNum == FirebaseJson::JSON_BOOL)
        Serial.println(jsonData->boolValue ? "true" : "false");
      else if (jsonData->typeNum == FirebaseJson::JSON_INT)
        Serial.println(jsonData->intValue);
      else if (jsonData->typeNum == FirebaseJson::JSON_DOUBLE)
        printf("%.9lf\n", jsonData->doubleValue);
      else if (jsonData->typeNum == FirebaseJson::JSON_STRING ||
               jsonData->typeNum == FirebaseJson::JSON_NULL ||
               jsonData->typeNum == FirebaseJson::JSON_OBJECT ||
               jsonData->typeNum == FirebaseJson::JSON_ARRAY)
        Serial.println(jsonData->stringValue);
    }
  }
}
```

mobizt, 2020-08-16:
1.
As I post above you need (essential) to take care of the request response. Non block for server request in not make sense and useless. You can use async or non block for server side but the client should wait for complete response.

2.
It's not the web socket. it is the http request on secure port (443) sent to sever with the header that the connection should keep alive  and server should not close it after sent the request.  The connection is kept alive as long as both client and server are on the internet.

3. I have answered this in many opened issues and also wrote the comment in the stream examples for long time.

mobizt, 2020-08-16:
If you've found the unacceptable latency or network round trip delay in your device, the problem may be at the router or AP.

mars000, 2020-08-16:
thanks for your answers - they assist a great deal in my understanding how to best use your very useful library.