mobizt/Firebase-ESP32 #94: ENHANCEMENT

mars000, 2020-08-17:
I see that the current library uses deprecated method for authentication.
Any thoughts on integrating JSON Web Token (JWT) into the code - this seems the preferred supported model from Google moving forward ?

https://cloud.google.com/endpoints/docs/openapi/authenticating-users-firebase

mobizt, 2020-08-17:
I planned and currently develop that which not because obsolete auth method but I want to include the C[loud Firestore and Cloud Storage](https://github.com/mobizt/Firebase-ESP32/issues/83) which required the access or custom token.

The custom token creation/renewal and Cloud Storage are working fine unless memory leaks which I'm trying to fix it.

closed 2020-08-17 by mobizt

mars000, 2020-08-17:
That’s exciting. Any idea of timeframe before we can use it ?

mobizt, 2020-08-17:
I can't tell the schedule, it deepens on the free time from my works and time used for bugs fix.

mars000, 2020-08-17:
No worries.

 

I’m sure you are across it already but just in case you are not have you seen this library ? https://github.com/GoogleCloudPlatform/google-cloud-iot-arduino/blob/master/src/jwt.h

 

I assume this can be used to create the JWT for Firebase login ?

mobizt, 2020-08-18:
I seen it for long time but it can't use with other Firebase product and services.

That library creates the signed JWT token with SHA256 signing method which all Firebase products and services required only RS256 which this can be done by the help of SSL/TLS library like mbedTLS that is already in ESP32 core library (this kind of functions are not available in ESP8266).

I don't have any problem to create the encrypt signed token and it done already.

mars000, 2020-08-18:
Ok thanks for explaining.

I’m using ESP32 exclusively these days.

Perhaps you can share BETA library in github; and community can help with debugging. I’m keen to move to a proper supported interface with ESP32 and Firebase.

mobizt, 2020-08-18:
It cannot said that database secret is not proper method as it needed and only way to use in the App Script.

mobizt, 2020-08-18:
If you reply from email, please don't include the old message in the reply. It looks as junk message.

mars000, 2020-08-18:
I mean it from the fact that Google saying its deprecated and could be high risk in future will no longer work.

mars000, 2020-08-18:
Sorry😊

mobizt, 2020-08-18:
Database secret still in work as it requires in App Script and I'm sure Google still be support this as they support the App Script.

mobizt, 2020-08-18:
Thanks for interesting but I'm not intend to share undocumented and unfinish work because the new library API changes and not compatible with the old version.

mars000, 2020-08-18:
thank you - happy to help where I can. Look forward to the update ;-)