mobizt/Firebase-ESP32 #95: HELP GPRS

vladigg54, 2020-09-03:
Hi, 

I use ESP32, how can I use TinyGSM with sim800 with FirebaseESP32.h (not HTTP, HTTPS, GET, POST)

I want to make simple "set"

Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
Firebase.setBool(data, "sync", true);

mobizt, 2020-09-03:
If I understand correctly, you want to use GSM module with ESP32 and want to access Firebase with this library. 

The library use the core WiFi client to work and also supported the ethernet.

The network bridge as GSM module is not supported and it's not implemented in this library.

closed 2020-09-03 by mobizt

mide-X, 2021-01-13:
But is there a way around this problem?
Say, a Library that supports using the SIM module with firebase.

mobizt, 2021-01-13:
Use pocket WiFi with sim card is better.

The GSM/GPRS is outdated and fade out from telecommunication industry.