mobizt/Firebase-ESP32 #96: BUG: json.toString

Ferni492, 2020-09-03:
Good afternoon, I've been designing a project for a while now using this incredible library. First of all I wanted to say thank you.
The problem I found is centered around the command: ".toString". This error happens sometimes and from what I could identify it doesn't depend on the content of the Json. What happens ? I have a function that transforms the json into a string and performs some operations, I execute it very often, but once every so often it crashes the microcontroller and reboots.

To rule out the possibility that the json content is the cause of the problem, before executing the command ".toString" I am executing ".clear ()" and the problem occurs anyway. I attach this function:

```
FirebaseJson json;

bool firebase_post_payload(const String path) {
 String jsonStr;
 json.clear();                                          // I know that it makes no sense to empty the json but it is to justify that the content does not generate the error
 json.toString(jsonStr, true);
 Serial.println("==>'" + jsonStr + "'");  // Sometimes the ESP32 reboots before executing this line 

 /*Here I execute some tasks like posting the Json to Firebase using WiFi or GSM*/
}

```
What surprises me the most is that sometimes it works and sometimes it doesn't. It is the same function!

Normally another function is responsible for "completing" the json using the command ".add ()", I also attach the result of printing the json after passing it to a string (when the json.clear() is commented):

```
'{
    "Temp_1": 25,
    "BATTERY": "3.04",
   "time": "2020-09-03T20:21:06Z"
}'
```
After verifying that I was not using any type of unsigned data, I made the decision to process the backtrace returned by the ESP32, I attached the analysis:

[image]

Based on the last 6 lines of the analysis, I managed to trace in my program where the error was being generated. With some "Serial.print", I came to the conclusion that the problem occurs right in the command mentioned already before in the post.

IDE: Arduino 1.8.13
ESP32 Dev Module

If more information is needed, I have no problem publishing it, the program is really extensive and I didn't want to make a really long post. Anyway, the other processes do not use json or library dependencies simultaneously.
Thank you very much.

mobizt, 2020-09-04:
I did not found the bug that you mentioned.
The serial printing for your code are below
```
==>'{}'
==>'{}'
==>'{}'
==>'{}'
==>'{}'
==>'{}'
==>'{}'
==>'{}'
```
The decoded stacks showed nothing helpful for this case.
You need to separate the suspected code from other code and test it alone to make sure the problem causes. This to prevent the out of memory or stack overflow problem from other tasks.

You can post your test code that causes the problem here.

closed 2020-09-04 by mobizt

mobizt, 2020-09-04:
You can try the FirebaseJson examples [here](https://github.com/mobizt/Firebase-ESP32/tree/master/examples/FirebaseJson/examples).

If you still get the issues, please check your board configuration in Arduino IDE.