mobizt/Firebase-ESP32 #98: HELP: JsonObject gain size and cause device resets

Victorhlab, 2020-09-21:
My firebase database setup is such that say field temperature is an array type. 
With each new temperature reading the array is appended and uploaded to Firebase.
Currently it logs once per 10s. 

After investigation, i realized that the json object is stored on the device. Which is unwanted.
I want the database to contain all this info and not the device too. 
So i want the device to push a new value to a path, and then to forget about it. So that the database can save all this data. And not the device too. Is this possible?

After some time, the device (ESP32) gets slower and slower to mange the appending of each new value.
It is getting so slow that the device calls abort() on a core, and the device resets.

I have also tried to clear the firebaseJson object, but on firebase the array's previous index get deleted and the new one is added. So you would have something like { temperature[ 8 : "23"  ] }. Meaning the current index is 8, and the prev index (7 and below) does not exist.

mobizt, 2020-09-21:
Don't understand what you're trying to do.
You may do something wrong with your code.

I can't help to correct your code unless the issue related to the usage or library malfunction.

You may need to make a clear code without third party library, all data e.g. sensor data are simulated and test it until.  If you make sure this related to library not the out of free heap and stack overflow, you can post the your clear code and all error message here.

closed 2020-09-21 by mobizt

mobizt, 2020-09-22:
You also need to know that the Firebase server return the data that changed back to client as payload. 

This means when you push some data (json or any), you will get that data back that stored in Firebase Json object which the JSON and JSON array type data will pass to internal FirebaseJson and FirebaseJsonArray objects which user can retrieve this data later which is the function of Firebase Data object. This will take 2 times memory of your push data.  To prevent this, you need to use updateNodeSilent instead. You can use updateNodeSilent to the path that doesn't exist, the new path will created instead of random uid key in the push operation.  With update NodeSilent, the server will not send the payload response to client.

Victorhlab, 2020-09-22:
Is it possible to post data to firebase without a Firebase data object?
The issue is that the firebase data object grows over time on the ESP32.
After a while it takes the processor too long to append array values that
is inside of the firebase dataobject.
So having a firebasedata object as a local copy on the embedded device is
not a good idea, since it can run out of memory with bigger datasets..
So i am thinking without a firebase dataobject, does firebase allow posting
data to firebase without having a local copy of the data on the device. I
dont want the embedded device to be aware of
the data. The application is a data logger

So im not  sure if I abusing firebase  - that firebase is meant for real
time only data, and not historic data that grows.

mobizt, 2020-09-22:
Did you read the first reply?
I can't help if you just mention something and nobody known what's about your coding.

mobizt, 2020-09-22:
If you want help, open the new issue with your testing code that causes the issues.

Describe what you are trying to do and what wrong with it.

Please respect my time and effort.