mobizt/Firebase-ESP32 #100: Floating Point Values in JSON

PrateekGoyal18, 2020-10-03:
I have some sensor data in float values. Is there some way by which I can add it to the JSON object?

mobizt, 2020-10-03:
Library has the built-in JSON builder and parser class.

Declare the **FirebaseJson** object and use the method **add** to add new value or use **set** to replace or create new value.

```C++

FirebaseJson json;

//Hard code
json.add("/node1", 123.456f);
//or
json.set("/node2", 0.123f); //this also creates the node2 if it does not existed.

//From variable
json.add("/node3", myVal); // myVal is the float variable. 

```
The supported variables are **bool**, **int**, **float**, **double**. 
The supported object are String, **FirebaseJson** and **FirebaseJsonArray**.
Unsupported variables are short, byte, uint8_t, uit16_t, uint32_t, long and so on.

Please check these [examples](https://github.com/mobizt/Firebase-ESP32/tree/master/examples) to learn how to use the FirebaseJson object.

closed 2020-10-03 by mobizt

PrateekGoyal18, 2020-10-03:
Getting this error:
C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\sketch\final_code.ino.cpp.o:(.literal._Z4loopv+0x40): undefined reference to `FirebaseJson& FirebaseJson::add<float>(String const&, float)'

C:\Users\Prateek\AppData\Local\Temp\ArduinoServer\MW\sketch\final_code.ino.cpp.o: In function `loop()':

C:\Users\Prateek\OneDrive\Documents\Arduino\SCD30 Sensor Setup\OLED\final_code/final_code.ino:237: undefined reference to `FirebaseJson& FirebaseJson::add<float>(String const&, float)'

collect2.exe: error: ld returned 1 exit status

mobizt, 2020-10-03:
You should update the library.

mobizt, 2020-10-03:
Recommend to update to v 3.8.0 (download is only available due to it updated 30 minutes ago)

PrateekGoyal18, 2020-10-03:
Thanks, @mobizt. It's working flawlessly now.