mobizt/Firebase-ESP32 #103: Using GSM module as Backup plan to connect Firebase

kavinsharma, 2020-10-10:
Hi @mobizt, I would really like to thank you for creating such a useful library,
I am thinking of interfacing the GSM module (SM 800) with ESP32 as a backup plan if wifi/internet connection is lost.
I was checking this [article ](https://randomnerdtutorials.com/esp32-sim800l-publish-data-to-cloud/)it uses the TinyGSM library to do an HTTP request. can we somehow use TinyGSM to connect to firebase and update values using the GSM module?

mobizt, 2020-10-11:
It's not supported and I also reply [here](https://github.com/mobizt/Firebase-ESP32/issues/97#issuecomment-695773736).

The 2G has no future and not available in many mobile operators worldwide.

closed 2020-10-11 by mobizt

Samorange1, 2021-05-24:
I dont know if this helps but I faced the same problem, however, I managed to solve it and I have posted my solution as a [GitHub Repo](https://github.com/Samorange1/Connect-ESP32-with-Firebase-via-GSM)