mobizt/Firebase-ESP32 #107: HELP

vladigg54, 2020-10-23:
I have error when compile. This error is from 3.8.1 or newer. If I use 3.8.0 or older version I have no error:

**From PlatformIO:**
In file included from lib/Firebase ESP32 Client/src/FirebaseESP32HTTPClient.h:43:0,
                 from lib/Firebase ESP32 Client/src/FirebaseESP32.h:52,
                 from src\main.cpp:9:
lib/Firebase ESP32 Client/src/wcs/WiFiClientSecureESP32.h:39:5: error: 'sslclient_context32' does not name a type
     sslclient_context32 *sslclient;
     ^

**From Arduino IDE:**
In file included from D:\DATA\Arduino\libraries\Firebase_ESP32_Client\src/FirebaseESP32HTTPClient.h:43:0,
                 from D:\DATA\Arduino\libraries\Firebase_ESP32_Client\src/FirebaseESP32.h:52,
                 from D:\DATA\Arduino\projects\esp32\ESP32_WiFiManager_FireBase\esp32_wifimanager_firebase_ver8\esp32_wifimanager_firebase_ver8\esp32_wifimanager_firebase_ver8.ino:13:
D:\DATA\Arduino\libraries\Firebase_ESP32_Client\src/wcs/WiFiClientSecureESP32.h:39:5: error: 'sslclient_context32' does not name a type
     sslclient_context32 *sslclient;
     ^
D:\DATA\Arduino\libraries\Firebase_ESP32_Client\src/wcs/WiFiClientSecureESP32.h: In member function 'int WiFiClientSecureESP32::socket()':
D:\DATA\Arduino\libraries\Firebase_ESP32_Client\src/wcs/WiFiClientSecureESP32.h:106:16: error: 'sslclient' was not declared in this scope
         return sslclient->socket = -1;
                ^

Regards!

mobizt, 2020-10-23:
Clean the installed Firebase library, and update to latest version (3.8.4 available via GitHub download).

vladigg54, 2020-10-23:
Same error with 3.8.4.

I found the problem, When I open example from Firebase-ESP32 library and just include "#include <HTTPUpdate.h>" without any other code and build - ERROR, when remove httpupdate - no error

Regards!

mobizt, 2020-10-23:
Ok, it's a bug when WiFiClientSecure.h included in your sketch or in some other included libs which I have created the modified version of WiFiClientSecureESP32 which uses the same processor directives and making the conflicts. 

I will update it soon.

vladigg54, 2020-10-23:
Thank you for fast response, I'll be waiting for update!

mobizt, 2020-10-23:
The bugs fixed, please update the library.

closed 2020-10-23 by mobizt