mobizt/Firebase-ESP32 #108: Stream cannot recover after internet restore

chauzov1995, 2020-10-30:
We simulate the situation: esp32 is connected to wi-fi, wi-fi has Internet access. We turn off the Internet on the router, then turn the Internet back on in a minute. The stream could not recover. There will be all the time - Stream timeout, resume streaming ...
Add a function to check if the stream is alive.

mobizt, 2020-10-30:
This is done internally and heavily discussion on this many times.

closed 2020-10-30 by mobizt

mobizt, 2020-10-30:
This is your internet connection issue.

Nothing to add for that but you can check the bool returns from **httpConnected()** method of Firebase Data object that you use for stream.

The **httpConnected()** will be true when stream connection is alive as the stream is connected (not the WiFi or internet status).

mobizt, 2020-10-30:
You can use **streamTimeout()** method of Firebase Data object that used for stream to check the stream connection too. 

That's the method use in the stream examples (not a stream callback), please check.