mobizt/Firebase-ESP32 #109: Support for ESP32-S2

atban98, 2020-10-31:
Please add support for the ESP32-S2 boards.

mobizt, 2020-10-31:
You can use ESP32-S2.
Did you have any issue?

closed 2020-10-31 by mobizt

atban98, 2020-11-02:
Hi mobizt,

I tried running the "Basic" example code on my ESP32-S2 and it fails at some point:

`16:09:17.095 -> Connecting to Wi-Fi........
16:09:19.940 -> Connected with IP: 172.20.10.5
16:09:19.940 -> 
16:09:19.940 -> ------------------------------------
16:09:19.940 -> Set double test...
16:09:21.393 -> PASSED
16:09:21.393 -> PATH: /Test/Double/Data1
16:09:21.393 -> TYPE: double
16:09:21.393 -> ETag: O2cc1ngwRgWFe1V3z3ctMRU2fuk=
16:09:21.393 -> VALUE: 10.123457000
16:09:21.393 -> ------------------------------------
16:09:21.393 -> 
16:09:21.664 -> PASSED
16:09:21.664 -> PATH: /Test/Double/Data2
16:09:21.664 -> TYPE: double
16:09:21.664 -> ETag: eeEpoYyyQzC1eIh6A+aBNgiNV1o=
16:09:21.664 -> VALUE: 20.123457000
16:09:21.664 -> ------------------------------------
16:09:21.664 -> 
16:09:21.901 -> PASSED
16:09:21.901 -> PATH: /Test/Double/Data3
16:09:21.901 -> TYPE: double
16:09:21.901 -> ETag: WuFQRCXZIOWLipRBydbf92QJNe4=
16:09:21.901 -> VALUE: 30.123457000
16:09:21.901 -> ------------------------------------
16:09:21.901 -> 
16:09:22.136 -> PASSED
16:09:22.136 -> PATH: /Test/Double/Data4
16:09:22.136 -> TYPE: double
16:09:22.136 -> ETag: KSX5GsUUnwCNb9Qg+vUSWxqtq60=
16:09:22.136 -> VALUE: 40.123457000
16:09:22.136 -> ------------------------------------
16:09:22.136 -> 
16:09:22.340 -> PASSED
16:09:22.340 -> PATH: /Test/Double/Data5
16:09:22.340 -> TYPE: double
16:09:22.340 -> ETag: i52HLZ9gqk5Cflpo+U+WW30CtbU=
16:09:22.340 -> VALUE: 50.123457000
16:09:22.340 -> ------------------------------------
16:09:22.373 -> 
16:09:22.541 -> PASSED
16:09:22.541 -> PATH: /Test/Double/Data6
16:09:22.541 -> TYPE: double
16:09:22.541 -> ETag: KNXbSDKY0tdg2wTGCEefCdFKWIY=
16:09:22.541 -> VALUE: 60.123457000
16:09:22.575 -> ------------------------------------
16:09:22.575 -> 
16:09:22.811 -> PASSED
16:09:22.811 -> PATH: /Test/Double/Data7
16:09:22.811 -> TYPE: double
16:09:22.811 -> ETag: 1ovID/E8fG0plelcpPrRdPL9QVc=
16:09:22.811 -> VALUE: 70.123457000
16:09:22.844 -> ------------------------------------
16:09:22.844 -> 
16:09:23.046 -> PASSED
16:09:23.046 -> PATH: /Test/Double/Data8
16:09:23.046 -> TYPE: double
16:09:23.046 -> ETag: 74GLfuelptHe66ekg5XGSi8Mon8=
16:09:23.046 -> VALUE: 80.123457000
16:09:23.046 -> ------------------------------------
16:09:23.046 -> 
16:09:25.446 -> PASSED
16:09:25.446 -> PATH: /Test/Double/Data9
16:09:25.446 -> TYPE: double
16:09:25.446 -> ETag: ZAB5RcMZYRqwqbbeEEQN5NJI6Tc=
16:09:25.480 -> VALUE: 90.123457000
16:09:25.480 -> ------------------------------------
16:09:25.480 -> 
16:09:25.684 -> PASSED
16:09:25.684 -> PATH: /Test/Double/Data10
16:09:25.684 -> TYPE: double
16:09:25.684 -> ETag: aQe4Ibp+YrMslPR+b64MIdrMCTo=
16:09:25.684 -> VALUE: 100.123457000
16:09:25.684 -> ------------------------------------
16:09:25.684 -> 
16:09:25.684 -> ------------------------------------
16:09:25.684 -> Get double test...
16:09:25.889 -> PASSED
16:09:25.889 -> PATH: /Test/Double/Data1
16:09:25.889 -> TYPE: double
16:09:25.889 -> ETag: O2cc1ngwRgWFe1V3z3ctMRU2fuk=
16:09:25.889 -> VALUE: 10.123457000
16:09:25.889 -> ------------------------------------
16:09:25.889 -> 
16:09:30.884 -> FAILED
16:09:30.884 -> REASON: read Timeout
16:09:30.918 -> ------------------------------------
16:09:30.918 -> 
16:09:36.834 -> FAILED
16:09:36.834 -> REASON: read Timeout
16:09:36.834 -> ------------------------------------
16:09:36.834 -> 
`

atban98, 2020-11-02:
Hi mobizt,

I tried running the "Basic" example code on my ESP32-S2 module and it fails at some point:

> 16:09:17.095 -> Connecting to Wi-Fi........
16:09:19.940 -> Connected with IP: 172.20.10.5
16:09:19.940 -> 
16:09:19.940 -> ------------------------------------
16:09:19.940 -> Set double test...
16:09:21.393 -> PASSED
16:09:21.393 -> PATH: /Test/Double/Data1
16:09:21.393 -> TYPE: double
16:09:21.393 -> ETag: O2cc1ngwRgWFe1V3z3ctMRU2fuk=
16:09:21.393 -> VALUE: 10.123457000
16:09:21.393 -> ------------------------------------
16:09:21.393 -> 
16:09:21.664 -> PASSED
16:09:21.664 -> PATH: /Test/Double/Data2
16:09:21.664 -> TYPE: double
16:09:21.664 -> ETag: eeEpoYyyQzC1eIh6A+aBNgiNV1o=
16:09:21.664 -> VALUE: 20.123457000
16:09:21.664 -> ------------------------------------
16:09:21.664 -> 
16:09:21.901 -> PASSED
16:09:21.901 -> PATH: /Test/Double/Data3
16:09:21.901 -> TYPE: double
16:09:21.901 -> ETag: WuFQRCXZIOWLipRBydbf92QJNe4=
16:09:21.901 -> VALUE: 30.123457000
16:09:21.901 -> ------------------------------------
16:09:21.901 -> 
16:09:22.136 -> PASSED
16:09:22.136 -> PATH: /Test/Double/Data4
16:09:22.136 -> TYPE: double
16:09:22.136 -> ETag: KSX5GsUUnwCNb9Qg+vUSWxqtq60=
16:09:22.136 -> VALUE: 40.123457000
16:09:22.136 -> ------------------------------------
16:09:22.136 -> 
16:09:22.340 -> PASSED
16:09:22.340 -> PATH: /Test/Double/Data5
16:09:22.340 -> TYPE: double
16:09:22.340 -> ETag: i52HLZ9gqk5Cflpo+U+WW30CtbU=
16:09:22.340 -> VALUE: 50.123457000
16:09:22.340 -> ------------------------------------
16:09:22.373 -> 
16:09:22.541 -> PASSED
16:09:22.541 -> PATH: /Test/Double/Data6
16:09:22.541 -> TYPE: double
16:09:22.541 -> ETag: KNXbSDKY0tdg2wTGCEefCdFKWIY=
16:09:22.541 -> VALUE: 60.123457000
16:09:22.575 -> ------------------------------------
16:09:22.575 -> 
16:09:22.811 -> PASSED
16:09:22.811 -> PATH: /Test/Double/Data7
16:09:22.811 -> TYPE: double
16:09:22.811 -> ETag: 1ovID/E8fG0plelcpPrRdPL9QVc=
16:09:22.811 -> VALUE: 70.123457000
16:09:22.844 -> ------------------------------------
16:09:22.844 -> 
16:09:23.046 -> PASSED
16:09:23.046 -> PATH: /Test/Double/Data8
16:09:23.046 -> TYPE: double
16:09:23.046 -> ETag: 74GLfuelptHe66ekg5XGSi8Mon8=
16:09:23.046 -> VALUE: 80.123457000
16:09:23.046 -> ------------------------------------
16:09:23.046 -> 
16:09:25.446 -> PASSED
16:09:25.446 -> PATH: /Test/Double/Data9
16:09:25.446 -> TYPE: double
16:09:25.446 -> ETag: ZAB5RcMZYRqwqbbeEEQN5NJI6Tc=
16:09:25.480 -> VALUE: 90.123457000
16:09:25.480 -> ------------------------------------
16:09:25.480 -> 
16:09:25.684 -> PASSED
16:09:25.684 -> PATH: /Test/Double/Data10
16:09:25.684 -> TYPE: double
16:09:25.684 -> ETag: aQe4Ibp+YrMslPR+b64MIdrMCTo=
16:09:25.684 -> VALUE: 100.123457000
16:09:25.684 -> ------------------------------------
16:09:25.684 -> 
16:09:25.684 -> ------------------------------------
16:09:25.684 -> Get double test...
16:09:25.889 -> PASSED
16:09:25.889 -> PATH: /Test/Double/Data1
16:09:25.889 -> TYPE: double
16:09:25.889 -> ETag: O2cc1ngwRgWFe1V3z3ctMRU2fuk=
16:09:25.889 -> VALUE: 10.123457000
16:09:25.889 -> ------------------------------------
16:09:25.889 -> 
16:09:30.884 -> FAILED
16:09:30.884 -> REASON: read Timeout
16:09:30.918 -> ------------------------------------
16:09:30.918 -> 
16:09:36.834 -> FAILED
16:09:36.834 -> REASON: read Timeout
16:09:36.834 -> ------------------------------------
16:09:36.834 ->

mobizt, 2020-11-02:
It's normal, read timeout means no server responses in time based on your internet connection status.