mobizt/Firebase-ESP32 #110: Add OnDisconnect Handler

Keegan-Cruickshank, 2020-11-01:
I am looking to add presence to my ESP32. A great way to do this is described here: 
https://firebase.googleblog.com/2013/06/how-to-build-presence-system.html

I think you are using something similar to their c++ library so I assume you could implement this:
https://firebase.google.com/docs/reference/cpp/class/firebase/database/disconnection-handler

I have of course thought of just using date refs and polling that every few mins but this method would be much cleaner.

I think this would be a great feature to allow users to know when their esp32 is online and offline with just a few lines of code.

mobizt, 2020-11-01:
The feature you mentioned is not possible in REST APIs (http) and it's only available on the supported platforms that Firebase SDK or Firebase Admin SDK was installed.

mobizt, 2020-11-01:
It's not pure C++ but requires the SDK to install in the supported platform not limited  resources and unsupported embedded device.

closed 2020-11-01 by mobizt

Keegan-Cruickshank, 2020-11-01:
Yes I understand, Thanks for the snappy reply. Might have to live with the date snapshot uploaded periodically. Just not sure of the scalability there.