mobizt/Firebase-ESP32 #111: Help Buffer Size ESP32

ggeo007, 2020-11-05:
Hello.
I'm struggling to find a way to set the BSSLBufferSize and ResponseSize using firebase on ESP32.
I've been using the FirebaseESP8266 library on my projects, and now I'm moving to ESP32.
Everything works fine on my project, but when I try to receive a large data from the server, ESP32 crashes, and I believe it happens because there is not enough allocated memory for this task (even when I have 150000+ of free Heap).
Is there any similar way to define a buffer size on firebaseESP32, as similar as setBSSLBufferSize and setResponseSize found on firebaseESP8266?

Thank you for your excelent work on this libraries!

mobizt, 2020-11-05:
No limit of buffer size as in ESP8266 which you should manage by yourself.

You should...

Avoid to get large payload by redesign your database.
Read smaller data multiple times.
Use query filter.

closed 2020-11-05 by mobizt

ggeo007, 2020-11-06:
When I try to read an object greater than 800 bytes I'm receiving this error. If I reduce the object's size to 700 bytes, esp32 reads perfectly. I have 156kb of free heap. Do you have any idea what's going on?

PC: 0x40001615
EXCVADDR: 0x40000000

Decoding stack results
0x400da783: FirebaseESP32::strpos(char const*, char const*, int) at C:\Users\geova\Documents\Arduino\libraries\Firebase_ESP32_Client\src\FirebaseESP32.cpp line 4991
0x400da7cd: FirebaseESP32::getHeader(char const*, char const*, char const*, int&, int) at C:\Users\geova\Documents\Arduino\libraries\Firebase_ESP32_Client\src\FirebaseESP32.cpp line 2740
0x400dab44: FirebaseESP32::parseRespPayload(char const*, server_response_data_t&, bool) at C:\Users\geova\Documents\Arduino\libraries\Firebase_ESP32_Client\src\FirebaseESP32.cpp line 2629
0x400de4aa: FirebaseESP32::handleResponse(FirebaseData&) at C:\Users\geova\Documents\Arduino\libraries\Firebase_ESP32_Client\src\FirebaseESP32.cpp line 3145
0x400deed8: FirebaseESP32::waitResponse(FirebaseData&) at C:\Users\geova\Documents\Arduino\libraries\Firebase_ESP32_Client\src\FirebaseESP32.cpp line 2900
0x400df145: FirebaseESP32::handleStreamRead(FirebaseData&) at C:\Users\geova\Documents\Arduino\libraries\Firebase_ESP32_Client\src\FirebaseESP32.cpp line 3678
0x400df1b9: FirebaseESP32::readStream(FirebaseData&) at C:\Users\geova\Documents\Arduino\libraries\Firebase_ESP32_Client\src\FirebaseESP32.cpp line 1961
0x400d638d: loop() at C:\Users\geova\Documents\Arduino\SmartHome_v2\central\esp32\central_v4.4\central_v4/central_v4.ino line 188
0x400e922d: loopTask(void*) at C:\Users\geova\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x40088ddd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

mobizt, 2020-11-06:
Don't expected anything from the stack trace without your code.

mobizt, 2020-11-06:
As I recommend above don't get large payload especially from stream.

Stream is for Realtime, small data only, you need to redesign your database.

Please look at this
```C++
void setStreamCallback(FirebaseData &fbdo, StreamEventCallback dataAvailablecallback, StreamTimeoutCallback timeoutCallback = NULL, size_t streamTaskStackSize = 8192);
```

The default stream stack size is 8k, you can increase this but reduce the payload data size is recommended.

That stack memory also use to store the mbedTLS resources while cpu services some interrupt from the system tasks which required large memory.

mobizt, 2020-11-12:
I finally found this is the bugs from the find string utilities which use internally caused this wdt reset issue. This happens to both ESP8266 and ESP32 Firebase libraries which the whole new code was written.

Now the bugs was fixed and I also add the **setResponseSize** function as in ESP8266 version.

Please update the library.

ggeo007, 2020-11-12:
Thank you so much for your effort!
 I'd decided to rewrite my code using Get functions instead of Stream functions. Now I will update my library and test everything again.
I really appreciate your work!