mobizt/Firebase-ESP32 #112: HELP - Firebase Timeout no matter what I do

Keegan-Cruickshank, 2020-11-10:
Hey there,

I have the following code. It was a part of a much larger sketch but I have taken out the firebase parts and refined them. No matter what I do, I cannot get firebase to connect. I did a few weeks back but then I moved to other parts of the sketch and when I came back it no longer works. If someone could take a 2 second look. I'm sure they will see what I am missing.

```arduino
#include <Arduino.h>
#include <FirebaseESP32.h>
#include <FirebaseJson.h>
#include <ArduinoJson.h>

StaticJsonDocument<200> doc;
FirebaseData firebaseData;

int RED = 0;
int BLUE = 0;
int GREEN = 0;

String esid = "Charge House 2.0";
String epass = "fairenough";

String location1 = "/lamps/";
String deviceID = "42272cac-ff77-4e6d-8dc0-320bca2378ef";

void streamCallback(StreamData data) {
  if(data.dataType() == "int") {
    if(data.dataPath() == "/color/r") {
      RED = data.intData();
    } else if (data.dataPath() == "/color/g") {
      GREEN = data.intData();
    } else if (data.dataPath() == "/color/b") {
      BLUE = data.intData();
    }
  } else if (data.dataType() == "json") {
    deserializeJson(doc, data.jsonString());
    RED = doc["color"]["r"];
    GREEN = doc["color"]["g"];
    BLUE = doc["color"]["b"];
  }
  Serial.println(RED);
  Serial.println(GREEN);
  Serial.println(BLUE);
}

void streamTimeoutCallback(bool timeout)
{
  if(timeout){
    Serial.println("Firebase Timeout");
  }
  delay(1000);
}

void setup() {
  Serial.begin(115200);
  WiFi.begin(esid.c_str(), epass.c_str());
  Firebase.begin("xxxxx-e4ff9.firebaseio.com", "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx");
  Firebase.setStreamCallback(firebaseData, streamCallback, streamTimeoutCallback);
  if (!Firebase.beginStream(firebaseData, "lamps/42272cac-ff77-4e6d-8dc0-320bca2378ef")){
    Serial.println(firebaseData.errorReason());
  }
}

void loop() {
  

}
```

The dev key has been removed for security but confirmed to be correct on my side. Here is a quick snapshot of my simple database:

[image]

Keegan-Cruickshank, 2020-11-10:
[image]

As you can see, this is my serial output

mobizt, 2020-11-11:
Check your internet connection.

In Arduino IDE, choose the example BearSSL https request and run it. It should ok if your internet is ok.

closed 2020-11-11 by mobizt

mobizt, 2020-11-11:
No need to use delay in the stream timeout callback.

The library try it best to resume by restart the WiFi and resume. 

If it fails to resume, your internet connection is the problem.

Keegan-Cruickshank, 2020-11-11:
Thanks for the tips. Turned out, with verbose logging, that it was a memory issue. Couldn't start pls with only 70kb ram. Using the NimBLE library instead of bluedroid solved my issue. I can now use Wi-Fi, bluetooth and firebase while still having heaps of memory.