mobizt/Firebase-ESP32 #113: HELP: How much heap should this take?

jshep321, 2020-11-12:
I have observed about 40k heap consumed (and freed up after .end) in my application.  Is this expected?  
(ESP32-WROOM32)

Thanks!
```
About to .end firebase
xPortGetFreeHeapSize :134324
After firebase.end
xPortGetFreeHeapSize :175096
````

mobizt, 2020-11-12:
It depends on the memory used by embedTLS and SSL client and other dynamic allocations, you should test it your self.

closed 2020-11-12 by mobizt

jshep321, 2020-11-12:
I did test per the above.  I was posting here to get some other inputs, but I guess I won't see them since this is closed already.

mobizt, 2020-11-12:
I already explain it above and you will not get the exactly free heap as there also used by system task unless you turn off all internal system tasks which the library depends on.

This is private repo which the opened issue should be a problem in usage or bugs or any improvement.

mobizt, 2020-11-12:
If you are wonder of the heap usage of 40k, this is normal as most of memory in the library used by mbedTLS and TCP client(lwip). The dynamic memory allocation is based on the response payload which always keep as long as no new http request.

You can test the amount of memory used by mbedTLS by using WiFiClientSecure to create https connection to some server (via port 443).