mobizt/Firebase-ESP32 #115: HELP Stream callback on resume

vladigg54, 2020-11-13:
Hi, I have a problem!

When I start ESP32 for the first time and successful boot, MCU check and read path in Firebase.setStreamCallback function. Everything is OK, but when I shutdown the router and ESP32 lost connectivity, MCU start timeout callback and print to serial "Stream timeout, resume streaming..." and again and again. In this moment I change key or value in this stream callback path and start again router. After router boot, ESP32 connect to WiFi end timeoutcallback print stop, but MCU does not check stream callback path like first boot and does not know for any change in this path.

Please help me!

mobizt, 2020-11-13:
It's possible block by your other code or delay if you don't use the stream callback.

Try with the stream example from the situation you test the router.

vladigg54, 2020-11-13:
I will check today and I will write result with example

vladigg54, 2020-11-13:
The problem is in my code. Everything is OK!

Thank You for fast response!

closed 2020-11-13 by vladigg54