mobizt/Firebase-ESP32 #117: BUG ESP32 Using BLE and Firebase same time causes memory error.

0100101101001011, 2020-11-28:
**Describe the bug**
When I try to use Arduino BLE and this library at the same time. The esp32 crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.9]

**ESP8266 Arduino Core SDK version**
 - Version [e.g. 2.5.1]

**Additional context**
Add any other context about the problem here.

mobizt, 2020-11-28:
Please search google How to use Bluetooth and WiFi at the same time?
It's currently not possible unless Espressif update the firmware.

closed 2020-11-28 by mobizt

mobizt, 2020-11-28:
If the Bluetooth and WiFi coexistent is possible, see the FAQ 5.3.2
https://www.espressif.com/sites/default/files/documentation/ESP32_FAQs__EN.pdf

It needs to config in menuconfig under ESP-IDF.

mobizt, 2020-11-28:
While Bluetooth is enable, please try the example in Arduino IDE examples about HTTPS request.
If you get the exception error, you can't use both WiFi and Bluetooth simultaneously as you don't have enough memory.