mobizt/Firebase-ESP32 #118: ESP32S2 warnings

poli44, 2020-12-02:
Hi, When compiling for ESP32S2 i got this. There is a chance to remove these warnings.

`In file included from /Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:40:
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:589:10: error: 'vector' in namespace 'std' does not name a template type
     std::vector<std::string> _pathTk = std::vector<std::string>();
          ^~~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:589:5: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:46:1:
+#include <vector>
 
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:589:5:
     std::vector<std::string> _pathTk = std::vector<std::string>();
     ^~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:590:10: error: 'vector' in namespace 'std' does not name a template type
     std::vector<eltk_t> _eltk = std::vector<eltk_t>();
          ^~~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:590:5: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
     std::vector<eltk_t> _eltk = std::vector<eltk_t>();
     ^~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:591:10: error: 'vector' in namespace 'std' does not name a template type
     std::vector<el_t> _el = std::vector<el_t>();
          ^~~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:591:5: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
     std::vector<el_t> _el = std::vector<el_t>();
     ^~~
In file included from /Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:40:
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:646:48: error: 'std::vector' has not been declared
     void _strToTk(const std::string &str, std::vector<std::string> &cont, char delim);
                                                ^~~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.h:646:54: error: expected ',' or '...' before '<' token
     void _strToTk(const std::string &str, std::vector<std::string> &cont, char delim);
                                                      ^
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp: In member function 'bool FirebaseJson::get(FirebaseJsonData&, const String&, bool)':
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:388:28: error: '_pathTk' was not declared in this scope
     _strToTk(path.c_str(), _pathTk, '/');
                            ^~~~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:388:28: note: suggested alternative: '_lastTk'
     _strToTk(path.c_str(), _pathTk, '/');
                            ^~~~~~~
                            _lastTk
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp: In member function 'size_t FirebaseJson::iteratorBegin(const char*)':
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:427:5: error: '_eltk' was not declared in this scope
     _eltk.clear();
     ^~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:430:5: error: '_el' was not declared in this scope
     _el.clear();
     ^~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:430:5: note: suggested alternative: '_nl'
     _el.clear();
     ^~~
     _nl
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp: In member function 'void FirebaseJson::iteratorEnd()':
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:438:5: error: '_eltk' was not declared in this scope
     _eltk.clear();
     ^~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp: In member function 'void FirebaseJson::iteratorGet(size_t, int&, String&, String&)':
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:450:9: error: '_eltk' was not declared in this scope
     if (_eltk.size() < index + 1)
         ^~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:458:9: error: '_eltk' was not declared in this scope
     if (_eltk[index].type == 0)
         ^~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp: In member function 'void FirebaseJson::_jsmn_parse(bool)':
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:503:5: error: '_eltk' was not declared in this scope
     _eltk.clear();
     ^~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp: In member function 'void FirebaseJson::_setMark(int, bool)':
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:538:28: error: '_el' was not declared in this scope
     for (size_t i = 0; i < _el.size(); i++)
                            ^~~
[... 293 lines cut ...]
                            ^~~~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:2239:28: note: suggested alternative: '_lastTk'
     _strToTk(path.c_str(), _pathTk, '/');
                            ^~~~~~~
                            _lastTk
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:2258:5: error: '_el' was not declared in this scope
     _el.clear();
     ^~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:2258:5: note: suggested alternative: '_nl'
     _el.clear();
     ^~~
     _nl
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:2259:5: error: '_eltk' was not declared in this scope
     _eltk.clear();
     ^~~~~
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp: In member function 'bool FirebaseJsonArray::_get(FirebaseJsonData&, const char*)':
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:2694:41: error: 'class FirebaseJson' has no member named '_pathTk'; did you mean '_lastTk'?
     _json._strToTk(path2.c_str(), _json._pathTk, '/');
                                         ^~~~~~~
                                         _lastTk
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp: In member function 'void FirebaseJsonArray::_set(const char*, const char*, bool)':
/Users/mariuszperkowski/Documents/Arduino/libraries/Firebase-ESP32-master/src/FirebaseJson.cpp:2785:33: error: 'class FirebaseJson' has no member named '_pathTk'; did you mean '_lastTk'?
     _json._strToTk(path2, _json._pathTk, '/');
                                 ^~~~~~~
                                 _lastTk`

mobizt, 2020-12-02:
Are you sure you set up all the IDE and Arduino ESP32 core SDK correctly?

This is the compiler error due to missing some tool chains.

closed 2020-12-02 by mobizt

poli44, 2020-12-02:
xtensa-esp32s2-elf-g++

mobizt, 2020-12-02:
The error due to the vector and it's already included [here](https://github.com/mobizt/Firebase-ESP32/blob/master/src/FirebaseJson.h#L44).

The lines at the error showed are different from the latest source of library. Do you use the outdated version or corrupted files?

Please update the library.

poli44, 2020-12-02:
I added  #include <vector> to file FirebaseJson.h and it help. Thanks.

poli44, 2020-12-02:
[image]

mobizt, 2020-12-02:
It different and my cause the conflicts and unexpected error due to you may ever installed the FirebaseJson library which may outdated and is not completely compatible with the one use in the Firebase library.

You should install the latest Firebase version.