mobizt/Firebase-ESP32 #120: Several different error messages on different hosts

MathiasYde, 2020-12-03:
Using a WEMOS LOLIN32 on the example basic Arduino sketch results in several different error messages. The microcontroller successfully connects to WiFi.

I've tried the following hosts:
- greenleaf-63767.europe-west1.firebaseio.com
  * bad request, verify the Firebase credentials and the node path
  * data type mismatch [on upload test]

- greenleaf-63767.firebaseio.com
  * data type mismatch, "404 Not Found
  * not found, "404 Not Found [with and after push integer test]

- greenleaf-63767-default-rtdb.europe-west1.firebasedatabase.app
  * bad request, verify the Firebase credentials and the node path

- greenleaf-63767.firebase.io
  * bad request, verify the Firebase credentials and the node path

mobizt, 2020-12-03:
You may currently read the large JSON payload from the server which causes the truncated payload and the error in internal JSON parsing process.

This because you never delete or minimize or limit the data under the path that being stream or get.

 You can set the size of SSL client rx/tx buffers in the case where you want to work with large data.
 firebaseData.setBSSLBufferSize(2048, 1024);

 And also set the size of HTTP response buffers to avoid the truncated response payload.
 firebaseData.setResponseSize(4096); 

The sufficient rx buffer size that can handle the SSL response is 16k bytes. Even the server may or may not support fragmentation.

Send and receive the small payload don't have the problem unless the large payload if server does not support fragmentation, the buffer need to increase to handle larger payload.

mobizt, 2020-12-03:
You may currently read the large JSON payload from the server which causes the truncated payload and the error in internal JSON parsing process.

The library default payload size limit is 4096 bytes which you can increase it to bigger value.

 firebaseData.setResponseSize(8192); 

I delete the previous comment due to it is for esp8266 version.

mobizt, 2020-12-03:
Please also check the valid of the Firebase credentials.

The error is on your code and database management.

Avoid to get or set the large JSON data as it not suitable in the limited resource device.

closed 2020-12-03 by mobizt

MathiasYde, 2020-12-03:
I've set the response size of firebaseData to 8192 and still get similar results. Regardless, how do I check the valid of my Firebase credentials?

mobizt, 2020-12-03:
Host name should valid no slash at the end.
Database secret should should not contain any space in side the quotes "".

MathiasYde, 2020-12-03:
I've made sure of that and still getting errors

mobizt, 2020-12-03:
If you change the node path to other node that contain small data e.g. only number or string inside and still get the error, try to create the new database secret.

If you still get the error, the problem may be the account restriction which I don't know.

Try to use other gmail to register the new project.

MathiasYde, 2020-12-03:
I'll use something else. Thank you.