mobizt/Firebase-ESP32 #121: HELP needed on the database size management

Sixonlon, 2020-12-08:
Hi there,

Thanks for your library, very usefull. I'm currently building a web server that read the data stored in a firebase database but I would like to manage it in that way : I want to limit the max number of data (actually I upload JSON data) stored to a certain number. When I need to upload new data, I would like the oldest one to be deleted and then add the new one. I had the idea to add in my JSON object the firebase timestamp and then querry my datas by using the timestamp. I would store the information and then use it to delete the oldest node before uploading a new one. 

I manage to upload my data with the timestamp but I'm stuck on the querry part. the way I structure my database is as followed : 
-key XXXXXXXXXX
      data : the actual data (an Integer)
      timestamp : the timestamp
-key YYYYYYYYYY
      data : the actual data (an Integer)
       timestamp : the timestamp
-key ZZZZZZZZZZ
   data : the actual data (an Integer)
   timestamp : the timestamp
....

Here is the code I use. What do I do wrong with the querry part ? Would there exist another way to do what I want ? I tough of generate a custom key instead of the random one generated automatically that name would be stored. I would then track the X latest name added and then delete the oldest one when I upload new data. But keeping track of that would be quite heavy 
in think.

Thanks in advance ;)
 
`#include <Arduino.h>
#include <WiFi.h>
#include <FirebaseESP32.h>

#define FIREBASE_HOST "XXXXXXXXb.firebaseio.com"
#define FIREBASE_AUTH "SECRET"
#define WIFI_SSID "WIFI_SSID "
#define WIFI_PASSWORD "WIFI_PASSWORD "

byte rxindex = 0;
char rxdata[32];

//Define Firebase Data object
FirebaseData firebaseData;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
}
void printResult(FirebaseData &data)
{
    if (data.dataType() == "int")
        Serial.println(data.intData());
    else if (data.dataType() == "float")
        Serial.println(data.floatData(), 5);
    else if (data.dataType() == "double")
        printf("%.9lf\n", data.doubleData());
    else if (data.dataType() == "boolean")
        Serial.println(data.boolData() == 1 ? "true" : "false");
    else if (data.dataType() == "string")
        Serial.println(data.stringData());
    else if (data.dataType() == "json")
[... 202 lines cut ...]

      process_command(rxdata);
      rxindex = 0;
    } else if (rxdata[rxindex] == '\b')
    { //backspace
      if (rxindex > 0)
      {
        Serial.print(F("\b \b"));
        rxindex--;
      }
    }
    else  //other char
    {
      Serial.print(rxdata[rxindex]);
      rxindex++;
      if (rxindex >= 32)
      {
        rxindex = 0;
      }
    }
  }
  // put the rest of the code here
}

`

mobizt, 2020-12-09:
You need to set the .indexOn in the database rules to the parent node with the name of child node to be queried.

Please see [this example](https://github.com/mobizt/Firebase-ESP32/issues/60).

closed 2020-12-09 by mobizt

Sixonlon, 2020-12-09:
Thks man!