mobizt/Firebase-ESP32 #122: How should I check if a object exists before saving it?

maylortaylor, 2020-12-14:
I am trying to get a simple user creation logic flow but can't quite seem to get the "does user exist" code to work correctly.

```
  String path = "users/" + String(WiFi.macAddress());
  FirebaseJson userJson;
  userJson.set("name", "New User");
  userJson.set("/createdDate/.sv", "timestamp");

  if (Firebase.get(firebaseData, path)) {
    if (Firebase.set(firebaseData, path, userJson)) {
      Serial.println("--------FB---USER CREATED .set() ----- ");
      Serial.println(firebaseData.dataPath());
      Serial.println(firebaseData.pushName());
      Serial.println(firebaseData.dataPath() + "/"+ firebaseData.pushName());

      printFirebaseResult(firebaseData);
      printFirebaseError();

      globalUser.id = firebaseData.pushName();
      globalUser.wifiName = globals.ssid;
      globalUser.timezone = globals.timeZone;
      globalUser.zipcode = weatherSettings.zipCode;

      globalUser.updateInitial = 0;

    } else {
      Serial.println(firebaseData.errorReason());
    }
  } else {
      Serial.println(firebaseData.errorReason());
  }
```
I am doing a `Firebase.get()` to see if the `path` exists. I thought this function returns `true` only if it has a `200 Response` and it finds something.

I've tested this multiple times and It will continuously keep creating this user (updating that `createDate` variable I made).
Am I doing this wrong?

I just want to create the user one time and then `on boot` of the ESP32, I can just check "does that user/object exist", if `yes` then don't create a user again.

mobizt, 2020-12-14:
If you want to check the node path exist, please use [pathExist](https://github.com/mobizt/Firebase-ESP32/tree/master/src#determine-whether-defined-database-path-exists-or-not) function. 

```C++
if(Firebase.pathExist(firebaseData, "path to check"))
{
  //node found
}
```

closed 2020-12-14 by mobizt

maylortaylor, 2020-12-14:
I tried this but I get a compilation error

```
error: 'class FirebaseESP32' has no member named 'pathExists'
   if (Firebase.pathExists(firebaseData, path)) {
```

mobizt, 2020-12-14:
Typo in your command pathExist not pathExist.
https://github.com/mobizt/Firebase-ESP32/tree/master/src#determine-whether-defined-database-path-exists-or-not