mobizt/Firebase-ESP32 #124: Patch fails if more than 2 key-value pairs

ishdemon, 2020-12-23:
```
void pushValues()
{
  Serial.println("push values");
  FirebaseJson updateData;
  updateData.set(TEMPS, t);
  updateData.set(HUMIDITY, h);
  updateData.set(VOLTAGE, voltage);
  updateData.set(POWER, power);
  updateData.set(TANK, tank_percent);
  updateData.set(SUMP, sump_percent);
  if (Firebase.updateNode(firebaseData2, sensor_path, updateData))
  {
    Serial.println(firebaseData2.dataPath());

    Serial.println(firebaseData2.dataType());

    Serial.println(firebaseData2.jsonString());
  }
  else
  {
    Serial.println(firebaseData2.errorReason());
  }
  //firebaseData2.stopWiFiClient();
}
```
This fails saying BAD REQUEST CHECK NODE PATH.. but when i only send two key - values it works. How to send all of them?? I already tried setwritesizelimit

mobizt, 2020-12-23:
Error is in your code which would be

1.  one or more of these variables i.e. **t, h, voltage, power, tank_percent** and **sump_percent** which is/are string contains some invalid characters which break the JSON structure.
2.  similar to 1, unless **TEMPS, HUMIDITY, VOLTAGE, POWER, TANK, SUMP** may contain invalid characters.
3.  sensor_path contains invalid token (string).

There no such error operation in this simple case and I already test your code.

closed 2020-12-23 by mobizt

mobizt, 2020-12-23:
You can serialize JSON to string like this

```cpp
String s;
updateData.toString(s);
Serial.println(s);
```
If the printed string is blank or invalid, there are characters that are not allow in the node value, You may need to use the escape char e.g. \\\ for \  and \\" for ".

ishdemon, 2020-12-24:
@mobizt Thanks. One sensor library was sending `Nan` as default. Bdw great library, just refactored my existing project which was using Blynk to Firebase. One thing i noticed that it kinda slows down my `SimpleTimer` tasks which works fine when used along Blynk. I guess HTTPS (Mbed TLS) is the reason. Does`EnableClassicRequest` will force it to use HTTP and make things a lot faster??

mobizt, 2020-12-24:
Firebase requires  secure connection using SSL/TLS encryption on port 443 and in others google cloud services.

The SSL client is not a problem only ssl handshake for the beginning of every new  connection takes time. If the  connection established, the consequence request can be faster as long as the tcp session is kept alive.

ESP32 has two cpu cores, the tasks can run independently based on priority that will interrupt low priority task. 

If the multiple socket connections were open (Firebase and Blynk) it shares the same WiFi radio and some connection can be interrupt or timed out.  Async SSL clients which open multiple sockets connections are not suitable for low memory device.

ishdemon, 2020-12-24:
> Firebase requires secure connection using SSL/TLS encryption on port 443 and in others google cloud services.
> 
> The SSL client is not a problem only ssl handshake for the beginning of every new connection takes time. If the connection established, the consequence request can be faster as long as the tcp session is kept alive.
> 
> ESP32 has two cpu cores, the tasks can run independently based on priority that will interrupt low priority task.
> 
> If the multiple socket connections were opened (Firebase and Blynk) they shared the same WiFi radio and some connection can be interrupt or timed out. Async SSL clients which open multiple sockets connections are not suitable for low memory device.

No i am not using Blynk and Firebase together..I just noticed the Timer tasks were not executing at defined intervals when i implemented with firebase.

mobizt, 2020-12-24:
I said the same thing. Don't reply with  old quoted message  as it costs the server data and we talk only two.

It depends on the timer library it may implement its own which may has the low priority. Use the hardware or os timer or change the library.