mobizt/Firebase-ESP32 #128: it will work on T8 V1.7 TTGO ESP32 4MB Flash. HELP

TechTSDHacking, 2020-12-27:
it will work on ttgo t8 v1.7
product link: https://www.amazon.in/Outgeek-Development-Board-Module-ESP32/dp/B07L3P1PXH/ref=sr_1_1?dchild=1&keywords=ttgo+t8&qid=1609076876&sr=8-1
AliExpress banned in our country.

mobizt, 2020-12-27:
Absolutely, as it is ESP32 based module.
The board you asked before has the modem, you can use it stand alone without touching the modem.

It's bad design to include sim800l on that. It should include 3G/4G enable module instead of 2G or gprs module (sim800l) which the technology is outdated.

TechTSDHacking, 2020-12-27:
ik but  ttgo t8 v1.7 will work for this library

mobizt, 2020-12-27:
Actually I don't clearly understand what you said.  It looks like spam.

The library works on all ESP32 based modules and TTGO v1.7 too. 

The ESP32 based module with built-in mobile modem that send AT command to onboard modem to connect the internet is not support unless you know how to operate it in WiFi only mode without using AT command and connect to the internet router by WiFi.

You need the one real board in hand and test it by yourself.

closed 2020-12-27 by mobizt

TechTSDHacking, 2020-12-27:
i want gprs connection to store data or read data witch will support this library

mobizt, 2020-12-27:
I always said not support any mobile modem (2G/gprs/3G/4G). Why you don't understand?.

TechTSDHacking, 2020-12-27:
so your library is fake

TechTSDHacking, 2020-12-27:
i need to report it

TechTSDHacking, 2020-12-27:
yhaaaahaaaaa

TechTSDHacking, 2020-12-27:
salle