mobizt/Firebase-ESP32 #129: HELP REGARDING DELAY

Murtaza7745, 2020-12-28:
I have configured my Realtime firebase and esp32 to toggle an led whenever esp32 pin is pulled low or high via switch connected to another gpio.
I want to add a scheduled task to check my connection with firebase server and a scheduled task to check my status of gpio connected to switch.

PLEASE HELP.
ITS URGENT.

mobizt, 2020-12-28:
I can't provide your code. You need to test it with yourself.

closed 2020-12-28 by mobizt

Murtaza7745, 2020-12-28:
I am facing a error of connection dropping while i add task scheduler to my code.
i need help for this issue.

mobizt, 2020-12-28:
Please post your code and I will check for any improper use.

Murtaza7745, 2020-12-28:
/**
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: k_suwatchai@hotmail.com
 * 
 * Github: https://github.com/mobizt
 * 
 * Copyright (c) 2020 mobizt
 *
*/

#include <WiFi.h>
#include <FirebaseESP32.h>
#include <TaskScheduler.h>

//1. Change the following info
#define WIFI_SSID ""
#define WIFI_PASSWORD ""
#define FIREBASE_HOST ""
#define FIREBASE_AUTH ""

//2. Define FirebaseESP8266 data object for data sending and receiving
FirebaseData fbdo;

#define led 2
#define sw 23

void t1Callback();
Task t1(1000, TASK_FOREVER, &t1Callback);

void t1Callback() {
    Serial.print("t1: ");
    Serial.println(millis());
    while (WiFi.status() != WL_CONNECTED)
    {
      Serial.print(".");
    }
}

Scheduler runner;

void setup()
{

  Serial.begin(115200);
  pinMode(led,OUTPUT);
  pinMode(sw,INPUT_PULLUP);
  digitalWrite(led,LOW);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  runner.init();
  Serial.println("Initialized scheduler");
  
  runner.addTask(t1);
  Serial.println("added t1");
  t1.enable();
  Serial.println("Enabled t1");

  //3. Set your Firebase info

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);

  //4. Enable auto reconnect the WiFi when connection lost
  Firebase.reconnectWiFi(true);

}

void loop()
{
  runner.execute();
  if(digitalRead(sw)==0){
      if(Firebase.setInt(fbdo, "/LED_Status",0))
      {
      //Success
    
        
        if(fbdo.intData()==0){
          digitalWrite(led,HIGH);
        }
    }
    else{
    //Failed?, get the error reason from fbdo

      Serial.print("Error in getInt, ");
      Serial.println(fbdo.errorReason());
    }
  }

  if(digitalRead(sw)==1){
     if(Firebase.setInt(fbdo, "/LED_Status",1))
      {
      //Success
    
        
        if(fbdo.intData()==1){
          digitalWrite(led,LOW);
        }
    }
    else{
    //Failed?, get the error reason from fbdo

      Serial.print("Error in getInt, ");
      Serial.println(fbdo.errorReason());
    }
  }
  if(Firebase.getInt(fbdo, "/LED_Status"))
  {
    //Success
    
    
    if(fbdo.intData()==1){
      digitalWrite(led,LOW);
    }
    else if(fbdo.intData()==0){
      digitalWrite(led,HIGH);
    }
    

  }else{
    //Failed?, get the error reason from fbdo

    Serial.print("Error in getInt, ");
    Serial.println(fbdo.errorReason());
  }
}

mobizt, 2020-12-28:
Your problem is the algorithm.

You access data as it is on the local memory all the time which so improper way to do. It is like you always press F5 on the web browser when visit some website which it takes time for page loading and costs your internet data usage.

I think you are new to Firebase.

For your case, you can use stream as in [this example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Stream_callback/Stream_callback.ino).

The stream connection is for listening the node value changes on the server, you don't need to call get function all the time. Server will push the data to device only when the data changed occurred on the server.

Check the basic examples project [here](https://medium.com/@vibrologic/serverless-iots-with-firebase-realtime-database-and-esp32-def049181b57).

Murtaza7745, 2020-12-29:
thanks for the above info. really helpful.
my question was, i want to use task scheduler to perform periodic task,
 while continuously polling the firebase db.

mobizt, 2020-12-29:
The stream run in the separately thread on core 1 of RTOS task. 

Your code and other libraries can't interrupt or delay the stream operation unless your internet connection is unstable.

You need to design your algorithm and debug when you integrate everything together.

First of all you need to familiar with the function and code you're working for the library you used and limitation of your device before integrated them together.

Murtaza7745, 2020-12-29:
ok, thanks alot.

pranavjasani, 2021-08-22:
hello @mobizt I reffered your code but it seems a bit confusing...

I am having same problem but I am using 8 gpio pins to control leds with firebase. I am facing the delay of nearly16 seconds. can you help me with this???

thank you

mobizt, 2021-08-22:
@pranavjasani

There are many possibilities e.g. your internet, your code and the low memory that leads to SSL handshake failure.

You should debug by printing free heap and something you want at any place in your code. 

Please try to disable or exclude third party library that do some blocking operation e.g. reading data from sensor or hardware.

Post simple, clean and complete code without third party libs that reproduced the issue and anyone can test.