mobizt/Firebase-ESP32 #131: BUG

iflyhere, 2020-12-29:
**Describe the bug**
On Cloud Messaging the ESP32 Rebooting when I call "Firebase.sendMessage()". I think it's a memory leak.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to your example [Cloud_Messaging.ino](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Cloud_Messaging/Cloud_Messaging.ino)
3. On "Firebase.sendMessage()"
4. Send Firebase Cloud Messaging...
[V][fb_ssl_client32.cpp:296] stop_ssl_socket(): Cleaning SSL connection.
[V][fb_ssl_client32.cpp:61] start_ssl_client(): Free internal heap before TLS 244500
[V][fb_ssl_client32.cpp:63] start_ssl_client(): Starting socket
[V][fb_ssl_client32.cpp:107] start_ssl_client(): Seeding the random number generator
[V][fb_ssl_client32.cpp:117] start_ssl_client(): Setting up the SSL/TLS structure...
[V][fb_ssl_client32.cpp:132] start_ssl_client(): Loading CA cert
[V][fb_ssl_client32.cpp:216] start_ssl_client(): Setting hostname for TLS session...
[V][fb_ssl_client32.cpp:233] start_ssl_client(): Performing the SSL/TLS handshake...
[V][fb_ssl_client32.cpp:259] start_ssl_client(): Verifying peer X.509 certificate...
[V][fb_ssl_client32.cpp:271] start_ssl_client(): Certificate verified.
[V][fb_ssl_client32.cpp:289] start_ssl_client(): Free internal heap after TLS 203212
[V][fb_ssl_client32.cpp:340] send_ssl_data(): Writing HTTP request...
[V][fb_ssl_client32.cpp:340] send_ssl_data(): Writing HTTP request...
CORRUPT HEAP: Bad tail at 0x3ffde03d. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4012e78f on core 1

**Firebase ESP32 Client**
 - Version [3.8.11]

**IDE and its version:**
 - PlatformIO
 - Core [5.0.3]
 - Home [3.3.1]

**ESP32 Arduino Core SDK version**
 - Version [(Git) 2.1.0+sha.a58a358]

mobizt, 2020-12-30:
update the library

mobizt, 2020-12-30:
I will check and inform later.

mobizt, 2020-12-30:
The bug was fixed, please update the library.

closed 2020-12-30 by mobizt