mobizt/Firebase-ESP32 #133: HELP

noppawitflam, 2021-01-10:
Hello I'm Noppawit, (Thai)
I need use firebase for my project but I use esp32s2 , Can you help me please?
My problem is esp32s2 can not connect firebase.
Thank you
Noppawit

mobizt, 2021-01-10:
Please post your code and the bebug message here.

noppawitflam, 2021-01-10:
`#include <WiFi.h>
#include "FirebaseESP32.h"

#define FIREBASE_HOST "esp32-btn-default-rtdb.firebaseio.com" //Change to your Firebase RTDB project ID e.g. Your_Project_ID.firebaseio.com
#define FIREBASE_AUTH "b2nkQTA6pk7fAAZWj8MHMs6H23x2PPgAR5tsnJsR" //Change to your Firebase RTDB secret password
#define WIFI_SSID "NOPPAWIT"
#define WIFI_PASSWORD "1212312121"

//Define Firebase Data objects
FirebaseData firebaseData1;
FirebaseData firebaseData2;

const int ledPin =  19; //GPIO19 for LED
const int swPin =  18; //GPIO18 for Switch
bool swState = false;
String path = "/Nodes";
String nodeID = "Node1"; //This is this node ID to receive control
String otherNodeID = "Node2"; //This is other node ID to control

void streamCallback(StreamData data)
{
  if (data.dataType() == "boolean") {
    if (data.boolData())
      Serial.println("Set " + nodeID + " to High");
    else
      Serial.println("Set " + nodeID + " to Low");
    digitalWrite(ledPin, data.boolData());
  }
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
  {
    Serial.println();
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
}

void setup()
{
  Serial.begin(115200);
  pinMode(ledPin, OUTPUT);
  pinMode(swPin, INPUT);
  Serial.println();
  Serial.println();
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
 
  if (!Firebase.beginStream(firebaseData1, path + "/" + nodeID))
  {
    Serial.println("Could not begin stream");
    Serial.println("REASON: " + firebaseData1.errorReason());
    Serial.println();
  }
  
  Firebase.setStreamCallback(firebaseData1, streamCallback, streamTimeoutCallback);
}

void loop()
{
  if (digitalRead(swPin) != swState) {
    bool _swState = swState;
    swState = digitalRead(swPin);

    if (Firebase.setBool(firebaseData2, path + "/" + otherNodeID, swState)) {
      if (swState)
        Serial.println("Set " + otherNodeID + " to High");
      else
        Serial.println("Set " + otherNodeID + " to Low");
    } else {
      swState = _swState;
      Serial.println("Could not set " + otherNodeID);
    }

  }
}`

[image]

[image]
c:/users/noppawit/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\Noppawit\Documents\Arduino\Hardware\espressif\esp32/tools/sdk/esp32s2/lib\librtc.a(rtc.o)(.text.rtc_pad_gpio_wakeup+0x50b7fb000000a9): could not decode instruction; possible configuration mismatch

mobizt, 2021-01-10:
The problem is no internet connection, your WiFi hotspot or router may not access to the internet.

mobizt, 2021-01-10:
You can test with other APs e.g. mobile hotspot.

Fantom-3000, 2021-01-10:
Добрый вечер! А правила в базе данных правильно выставлены? База данных в тестовом режиме?

mobizt, 2021-01-10:
@Fantom-3000 The database rules are not applied  when you authenticate with legacy token (database secret).

closed 2021-01-11 by mobizt