mobizt/Firebase-ESP32 #135: HELP!!! To Connect with Firebase via SIM800L

mide-X, 2021-01-13:
I'm working with ESP32 Sim800L board and was wondering if there's a way to connect with firebase through a Sim module?

mobizt, 2021-01-13:
It's not support any network bridge which already stated in [readme](https://github.com/mobizt/Firebase-ESP32#unsupported-mobile-network-modem-bridge).

closed 2021-01-13 by mobizt

giselybelloadur, 2021-10-04:
> It's not support any network bridge which already stated in [readme](https://github.com/mobizt/Firebase-ESP32#unsupported-mobile-network-modem-bridge).

Can any modules connect to firebase, SIM800L, or others? Any suggestions on how to make this connection indirectly?

mobizt, 2021-10-05:
@giselybelloadur It stated clear unsupported by the library.

This is GitHub repo for issue and help related to the library, other question should ask on community forum and web board.