mobizt/Firebase-ESP32 #136: streamTimeoutCallback doesn't trigger when there is SSL error.

0100101101001011, 2021-01-14:
**Describe the problem**
I am testing the stream callback code. I noticed if there is a problem with the SSL connection, the streamTimeoutCallback does not get called. Following is the serial output during the SSL connection error.

**To Reproduce**
```
[E][fb_ssl_client32.cpp:98] start_ssl_client(): Connect to Server failed!
[E][FB_WCS32.cpp:160] connect(): start_ssl_client: -1
```

mobizt, 2021-01-14:
The stream timeout still working correctly but the timeout callback will be notified at some interval time since the stream connection failed to prevent too much notify status printed on the serial.

mobizt, 2021-01-14:
This is the library design.

closed 2021-01-14 by mobizt

mobizt, 2021-01-14:
Don't use the stream timeout callback to perform your work. 
It design for notification at some interval only.

You should use fbdo.httpConnected(); which returns true when the server connection established and should return true as long as the stream connection is keep alive and return false when the stream timed out or connection disconnected.

0100101101001011, 2021-01-18:
Thank you for helping! This library is amazing:)