mobizt/Firebase-ESP32 #137: HELP MultiPathStreamCallback not working if database grows

Rakado93, 2021-01-17:
I am setting a multipathstream call back which works fine for a small database. However when the database grows (outside the targeted node) the ESP32 doesn't responds to changes in the targeted node any more. Thank you for any help to understand the problem.

mobizt, 2021-01-17:
You may need to set stack size
https://github.com/mobizt/Firebase-ESP32/tree/master/src#set-the-multiple-paths-stream-callback-functions

And set the response size
https://github.com/mobizt/Firebase-ESP32/tree/master/src#set-the-http-response-size-limit

Please read this issue for limitation and recommendation. https://github.com/mobizt/Firebase-ESP32/issues/89

closed 2021-01-17 by mobizt

mobizt, 2021-01-18:
Normally the large payload of data under the stream path is unsuitable for the limited memory device.

Any changes in the parent node i.e. its child node changes, the stream event data of that changes will be pushed to the device regardless of whether that node path included in the child path array assigned with the function **beginMultiPathStream** or not.

Set the response size of my above comment can help but you may need to redesign your database based on the memory perspective.

Stream data should be kept as small as possible due to it designed for realtime data. The large log or history data should not included in the stream path.