mobizt/Firebase-ESP32 #138: HELP REGARDING MULTIPATH STREAM

Murtaza7745, 2021-01-25:
I am using esp32 to read and write three node from RTDB

When a multipath callback occurs, is there a way to get only the path whose value has changed?

Awaiting reply.

mobizt, 2021-01-25:
It is what you expected unless when the stream timed out, server will return all data under the parent path once when the stream is reconnected. This data will be processed as stream data and also trig the callback event.

You can use the stream time out callback to get the timed out status and store in some boolean variable. 

Then ignore the first coming stream event data if the timed out status was true.  The consequence stream data will be the actual changes in your database.

Murtaza7745, 2021-01-25:
i cant understand, can you please elaborate?

mobizt, 2021-01-25:
When the stream was established or reconnected, the stream data from readStream and stream callback function will be whole under  the parent path or stream path which is not the data that changes in some child nodes.

You may need to ignore this **first stream data** and process any consequence stream data.

Please check the git below, I modified it from the multipath stream example to ignore the **first stream data** that triggered after the stream is establish or reconnected (resumed from timed out).
https://gist.github.com/mobizt/d6c311af02f623d5df96c97a8932990f

closed 2021-01-25 by mobizt

mobizt, 2021-01-25:
I may miss what you're asking.

If you only want to get the event path, [check this](https://gist.github.com/mobizt/d6c311af02f623d5df96c97a8932990f#file-esp32_multipath_stream-ino-L59).

Murtaza7745, 2021-01-25:
Its helpful.
See, I'm changing three node values at the same time using a single button from an android app.
but even the above code is reading the last value only in esp32.

I want it to read all the values that have been updated not just the last one.

mobizt, 2021-01-25:
It's not like MQTT and the web socket used in other platforms that required Firebase SDK or Firebase Admin SDK to be installed.

The library works with REST APIs (http) which is the simple sending http request and wait to get response.

For stream mode, http connection is kept alive and device is waiting for the event data payload push from server without any further request unless the connection lost.

No guarantee that all server changes will be pushed to the client, it depends on server side operation, connection status during the remote data changes. It can't cache any change as MQTT do during client offline.

Murtaza7745, 2021-01-25:
so there is no alternative to this issue?

mobizt, 2021-01-25:
No this is the technology limit, not an issue.

I also comment about this in other issues too.

mobizt, 2021-01-25:
You should use MQTT with own database as it meets your requirement.

Murtaza7745, 2021-01-25:
ok thanks for the advice. really helpful.