mobizt/Firebase-ESP32 #139: ENHANCEMENT

Niltonsf, 2021-01-26:
Hello,

I have been trying to work with your ESP32 library with Firestore. I'm trying to set up the firebase_host, api_key, etc. But I'm having problems finding the right api_key, and the FIREBASE_HOST location. My suggestion is, could you please inform the location from where to find FIREBASE_HOST and API_KEY in the documentation.

I have a quick question. If I set up the firebase rules for only reading, writing if authenticate. Does your library will work with that?

Thanks, wish you the best.

mobizt, 2021-01-26:
The [basic examples](https://github.com/mobizt/Firebase-ESP32#basic-examples) in README show how to setup.

closed 2021-01-26 by mobizt

Niltonsf, 2021-01-26:
I have checked now I got it, will your library work for Firestore?

mobizt, 2021-01-27:
It's not support Firestore now.
https://github.com/mobizt/Firebase-ESP32/issues/84
https://github.com/mobizt/Firebase-ESP32/issues/47

mobizt, 2021-01-27:
For better performance, I choose gRPC instead of REST API

I'm looking for gRPC support from espresif.
https://github.com/espressif/esp-idf/issues/3352

mobizt, 2021-01-27:
Espressif release some ESP-VA-SDK which using gRPC and HTTP2 but it close source.
https://www.espressif.com/en/news/Google_Dialogflow_with_ESP32
https://github.com/espressif/esp-va-sdk

mobizt, 2021-01-27:
Firesstore is not support API_KEY as you mentioned, it support only access or OAuth2.0 token.

mobizt, 2021-01-27:
The Cloud Firestore via REST API will be added support to the new [Firebase ESP Client library](https://github.com/mobizt/Firebase-ESP-Client) in the next update.

The Firestore only support the id token using Email/password or custom JWT token and OAuth2.0 token for authentication.

The library supports these authentication methods.

Set the security rules for any function condition that allows public access other than **request.auth** is not secured.

mobizt, 2021-01-29:
The [Firebase ESP Client library](https://github.com/mobizt/Firebase-ESP-Client) with Cloud Firestore supports is available.

Niltonsf, 2021-01-29:
@mobizt Hello, I'm trying to use the new Firestore library, but I'm getting the following error when attempting to upload the code:
```
Multiple libraries were found for "WiFi.h"
 Used: /Users/niltonsf/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
In file included from /var/folders/05/spv322w148v8nv75jdgv13th0000gn/T/arduino_modified_sketch_140502/sketch_jan29a.ino:10:0:
/Users/niltonsf/Documents/Arduino/libraries/Firebase-ESP-Client-main/src/Firebase_ESP_Client.h:38:26: fatal error: rtdb/FB_RTDB.h: No such file or directory
 #include "rtdb/FB_RTDB.h"
                          ^
compilation terminated.
 Not used: /Users/niltonsf/Desktop/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /Users/niltonsf/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /Users/niltonsf/Desktop/Arduino.app/Contents/Java/libraries/SD
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.
```

mobizt, 2021-01-29:
You should uninstall or remove SD.h library from third party because this SD.h is already exist in core library.

The SD.h from third party is not compatible with SD.h in core library, my library use the standard function of SD.h in core library.

When compiler choose the SD.h from third party instead of SD.h in the core library, the conflicts occurred and can't compile.

Niltonsf, 2021-01-29:
@mobizt Should I remove the WiFi and SD from the esp32/1.0.4/libraries? or from the Arduino library?

This other error is due to the WiFi and SD:
```
In file included from /var/folders/05/spv322w148v8nv75jdgv13th0000gn/T/arduino_modified_sketch_408802/sketch_jan29a.ino:21:0:
/Users/niltonsf/Documents/Arduino/libraries/Firebase-ESP-Client-main/src/Firebase_ESP_Client.h:38:26: fatal error: rtdb/FB_RTDB.h: No such file or directory
 #include "rtdb/FB_RTDB.h"
 ```

mobizt, 2021-01-29:
Remove only SD.h, the compiler breaks at that line and compiler reports other errors that are not accurate and not necessary.

mobizt, 2021-01-29:
This issue  caused by you install the Arduino ESP32 Core SDK (tool chains and compiler) from Board Manager in Arduino IDE and from GitHub clone.

Which you need to choose only one installation method instead of install both methods as compiler found the same file appear in many locations and shows the errors.

Niltonsf, 2021-01-29:
@mobizt I have removed both libraries and I'm not getting that error anymore. But I'm still getting this:
```
In file included from /Users/niltonsf/Documents/Arduino/libraries/Firebase-ESP-Client-main/examples/Firestore/Create_Documents/Create_Documents.ino:21:0:
/Users/niltonsf/Documents/Arduino/libraries/Firebase-ESP-Client-main/src/Firebase_ESP_Client.h:38:26: fatal error: rtdb/FB_RTDB.h: No such file or directory
 #include "rtdb/FB_RTDB.h"
 ```

mobizt, 2021-01-29:
Please read my comment above, you make the conflicts in the installation of Arduino ESP32 core SDK from two sources.

Niltonsf, 2021-01-29:
@mobizt I should remove the ```https://dl.espressif.com/dl/package_esp32_index.json```from Additional Boards manager urls? Because I attempted to find another location for esp32 library but there is only one installed, the GitHub one is not installed.

mobizt, 2021-01-29:
From this repos https://github.com/espressif/arduino-esp32#installation-instructions

You should choose only the first method instead or second method of install both methods.

First method is recommended.
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

If you already install using the second method like this 
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md

You need to delete the folder /Documents/Arduino/hardware/espressif/esp32

Niltonsf, 2021-01-29:
I have removed the library esp32 from Espressif, but I keep getting this:
```
WARNING: Error loading hardware folder /Users/niltonsf/Documents/Arduino/hardware/espressif
  No valid hardware definitions found in folder espressif.
```
I have added the first method and installed esp32 again with the new method

Niltonsf, 2021-01-29:
@mobizt I get the following error when trying to compile from this repository:
```
Traceback (most recent call last):
  File "esptool.py", line 57, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports.py", line 29, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_posix.py", line 31, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_osx.py", line 32, in <module>
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
Failed to execute script esptool
exit status 255
/Users/niltonsf/Desktop/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board DOIT ESP32 DEVKIT V1
```
The repo I used was:
```
https://dl.espressif.com/dl/package_esp32_index.json
```

mobizt, 2021-01-30:
Your installation may not complete due to missing some dependencies files.

You should  reinstall Arduino IDE and install ESP32 Core SDK using this method again.
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

Niltonsf, 2021-01-31:
@mobizt I don't think that is a wrong installation, since I have done 2 fresh reinstalls and the library still gives errors

mobizt, 2021-01-31:
You need to update the latest library due to v 1.1.3 missing source files.

mobizt, 2021-01-31:
Sorry for this inconvenient because the files refactor.

Niltonsf, 2021-01-31:
@mobizt 
Hey, Where is the Firestore part in the new update. Haven`t seen any example with the name firestore.