mobizt/Firebase-ESP32 #142: HELP

Niltonsf, 2021-01-31:
@mobizt 
Hello, I'm trying to work with your new Firestore library.
I'm attempting to a Create_Document, I haven't changed any info for creating the document, I have only inserted WIFI info, Authentication info and the firebase_id.
```
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "MySSid"
#define WIFI_PASSWORD "MyPassword"

/* 2. Define the project ID */
#define FIREBASE_PROJECT_ID "automacao-com-aplicativo"

/* 3. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "test@test.com"
#define USER_PASSWORD "128236"

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long dataMillis = 0;
int count = 0;

void setup()
{

    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    /* Assign the user sign in credentials */
    auth.user.email = USER_EMAIL;
    auth.user.password = USER_PASSWORD;

    Firebase.begin(&config, &auth);
    Firebase.reconnectWiFi(true);

#if defined(ESP8266)
    //Set the size of WiFi rx/tx buffers in the case where we want to work with large data.
    fbdo.setBSSLBufferSize(1024, 1024);
#endif
}

void loop()
{

    if (millis() - dataMillis > 60000 || dataMillis == 0)
    {
        dataMillis = millis();

        String content;
        FirebaseJson js;
        
        //We will create the nested document in the parent path "a0/b0/c0
        //a0 is the collection id, b0 is the document id in collection a0 and c0 is the collection id id in the document b0.
        //and d? is the document id in the document collection id c0 which we will create.
        String documentPath = "a0/b0/c0/d" + String(count);

        js.set("fields/count/integerValue", String(count).c_str());
        js.set("fields/status/booleanValue", count % 2 == 0);
        js.toString(content);
        
        count++;

        Serial.println("------------------------------------");
        Serial.println("Create a document...");

        if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "test_collection" /* databaseId can be (default) or empty */, documentPath.c_str(), content.c_str()))
        {
            Serial.println("PASSED");
            Serial.println("------------------------------------");
            Serial.println(fbdo.payload());
            Serial.println("------------------------------------");
            Serial.println();
        }
        else
        {
            Serial.println("FAILED");
            Serial.println("REASON: " + fbdo.errorReason());
            Serial.println("------------------------------------");
            Serial.println();
        }
    }
}
```
This is where I got the project id:
[image]

mobizt, 2021-01-31:
Thank you for informing the issue. 

The problem is the API key is missing from all Firestore examples.

Please see the RTDB examples about the API key  assignment.

The Email, Password, API Key are the credentials for id token generation.

There is no incomplete or missing token error info in Firestore functions right now and I will fix this in the next update.

I'm apologize for this inconvenient.

Niltonsf, 2021-01-31:
@mobizt 
Oh ok.
This:
```
/* 2. Define the Firebase project host name and API Key */
#define FIREBASE_HOST "PROJECT_ID.firebaseio.com"
#define API_KEY "API_KEY"
``` 
into my code? Should I only insert this part in the Firestore's examples?

Niltonsf, 2021-01-31:
@mobizt 
I have added:
```
#define FIREBASE_HOST "PROJECT_ID.firebaseio.com"
#define API_KEY "API_KEY"

void setup(){
    config.host = FIREBASE_HOST;
    config.api_key = API_KEY;
}
```
I could successfully create the document, great work!

Niltonsf, 2021-02-01:
@mobizt 
Could you help me accessing the fbdo.payload().
I have attempted doing what is written but I might be doing it wrong:
This is what I receive:
```
Get a document...
PASSED
------------------------------------
{
  "name": "projects/automacao-com-aplicativo/databases/(default)/documents/user_data/MwAeYz8RnlQdTf4zqxV7gxOJREg2/Buttons/botao1",
  "fields": {
    "nome_botao": {
      "stringValue": "botao1"
    },
    "value": {
      "booleanValue": true
    },
    "nome": {
      "stringValue": "Quarto"
    },
    "icon": {
      "stringValue": "LightBulb"
    }
  },
  "createTime": "2021-02-01T17:59:56.695439Z",
  "updateTime": "2021-02-01T18:13:49.703817Z"
}
``` 
I have attempted doing this:
```
if (Firebase.Firestore.getDocument(&fbdo, FIREBASE_PROJECT_ID, "", documentPath.c_str(), mask.c_str()))
        {
            Serial.println("PASSED");
            Serial.println("------------------------------------");
            Serial.println(fbdo.payload());
            json = fbdo.jsonObject();
            json.get(jsonData, "value");
            Serial.println(jsonData.type);
            Serial.println("------------------------------------");
            Serial.println();
        }
```
I want to get the Boolean value from the result I received.

mobizt, 2021-02-01:
The payload response is not set to the internal FirebaseJson of Firebasse Data object as in the RTDB right now. 

At this time you can do external parse like this.
```cpp

  FirebaseJson json;
  FirebaseJsonData jsonData;

  json.setJsonData(fbdo.payload());

  json.get(jsonData, "fields/value/booleanValue");

  Serial.println(jsonData.stringValue);

```

closed 2021-02-02 by mobizt