mobizt/Firebase-ESP32 #143: HELP

semenesp32, 2021-02-02:
@mobizt
Hello, I'm trying to work with your new Firesbase-ESP32 library. I tried to create a new file using your library, but I got errors when compiling the file. I'm using a WEMOS LOLI32 board, Arduino ide version 1.8.5.
File code:
#include <WiFi.h>
#include <FirebaseESP32.h>
#include "DHTesp.h"
#include<stdlib.h>
 
#define DHTpin 15
DHTesp dht;
 
char buff[10];
#define FIREBASE_HOST "https://esp32andfirebase.firebaseio.com/"
#define FIREBASE_AUTH "gga1JXWY5rgyBHk56RSXLn3FPpajWfcq6itIM3nI"
#define WIFI_SSID "fawadkhan"
#define WIFI_PASSWORD "fawadkhan12"
 
 
//Define FirebaseESP32 data object
FirebaseData firebaseData;
FirebaseJson json;
 
 
String mytemp; 
String myhum;
void setup()
{
 
   Serial.begin(115200);
   dht.setup(DHTpin, DHTesp::DHT11);
   WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();
 
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
 
  //Set database read timeout to 1 minute (max 15 minutes)
  Firebase.setReadTimeout(firebaseData, 1000 * 60);
  //tiny, small, medium, large and unlimited.
  //Size and its write timeout e.g. tiny (1s), small (10s), medium (30s) and large (60s).
  Firebase.setwriteSizeLimit(firebaseData, "tiny");

    Serial.println("------------------------------------");
  Serial.println("Connected...");
}
 
void loop()
{
   delay(dht.getMinimumSamplingPeriod());
 
  float humidity = dht.getHumidity();
  float temperature = dht.getTemperature();
  myhum = dtostrf(humidity,3,2,buff);
  mytemp = dtostrf(temperature,3,2,buff);
  Serial.println(myhum);
  Serial.println(mytemp);
   
delay(100); 
 json.set("/hum", myhum);
 json.set("/temp", mytemp);
 Firebase.updateNode(firebaseData,"/Sensor",json);
}

Errors:
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'bool FirebaseESP32::parseSAFile()':
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:273:44: error: 'class String' has no member named 'clear'
             _cfg->signer.data->stringValue.clear();
                                            ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'bool FirebaseESP32::handleResponse(FirebaseData&)':
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:4826:38: error: 'class String' has no member named 'clear'
           fbdo._jsonData.stringValue.clear();
                                      ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'void FirebaseESP32::handlePayload(FirebaseData&, server_response_data_t&, char*)':
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:4943:30: error: 'class String' has no member named 'clear'
   fbdo._jsonData.stringValue.clear();
                              ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:4964:10: error: 'class String' has no member named 'clear'
     test.clear();
          ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:4976:10: error: 'class String' has no member named 'clear'
     test.clear();
          ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:4988:11: error: 'class String' has no member named 'clear'
   rawJson.clear();
           ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:4989:10: error: 'class String' has no member named 'clear'
   rawArr.clear();
          ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'void FirebaseData::clear()':
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:7442:25: error: 'class String' has no member named 'clear'
   _jsonData.stringValue.clear();
                         ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'void StreamData::empty()':
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:8094:28: error: 'class String' has no member named 'clear'
     _jsonData->stringValue.clear();
                            ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'void FCMObject::fcm_preparePayload(std::__cxx11::string&, fb_esp_fcm_msg_type)':
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:8487:5: error: 'class String' has no member named 'clear'
   s.clear();
     ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'bool MultiPathStreamData::get(const String&)':
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:8544:9: error: 'class String' has no member named 'clear'
   value.clear();
         ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:8545:8: error: 'class String' has no member named 'clear'
   type.clear();
        ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:8546:12: error: 'class String' has no member named 'clear'
   dataPath.clear();
            ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp: In member function 'void MultiPathStreamData::empty()':
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:8612:12: error: 'class String' has no member named 'clear'
   dataPath.clear();
            ^
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:8613:9: error: 'class String' has no member named 'clear'
   value.clear();
C:\Users\79084\Documents\Arduino\libraries\Firebase-ESP32-master\src\FirebaseESP32.cpp:8614:8: error: 'class String' has no member named 'clear'
   type.clear();
The same errors occur when I try to compile any of your example files. 
I would appreciate your help.

mobizt, 2021-02-02:
The library installation problem.

Please delete the library from the libraries folder and install again.

mobizt, 2021-02-02:
When you get the compilation error, you should post all errors instead of some part.

mobizt, 2021-02-02:
Look like you are not yet install ESP32 core SDK or incomplete install or out dated.

Please install the latest Arduino ESP32 Core SDK.

https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

closed 2021-02-02 by mobizt

semenesp32, 2021-02-02:
I have updated the ESP32 Core SDK. Everything worked. Thank you very much for your help.