mobizt/Firebase-ESP32 #145: HELP

Niltonsf, 2021-02-05:
Hello, how you doing? @mobizt 

I have been working with your library and it is working great for now. I just came across an issue I believe. I have created a program with the user sends the USER_EMAIL and the USER_PASSWORD from the bluetooth serial. I manage to get the data and use with the the auth user.email and password. The issue I'm having is, when I attempt to use:
```
 Firebase.begin(&config, &auth);
    Firebase.reconnectWiFi(true);
```
in a function or anywhere in the code I get this error:
```
Writing at 0x0009c000... (81 %)Sketch uses 1499982 bytes (114%) of program storage space. Maximum is 1310720 bytes.text section exceeds available space in board

Global variables use 63668 bytes (19%) of dynamic memory, leaving 264012 bytes for local variables. Maximum is 327680 bytes.
```
My function is called once only in the main page according to my logic, and whenever the arduino RESETs it will be called again:
```
  }
  
  //Cria o nome do esp dentro do Banco de dados
  char email_char[email.length()] = "", password_char[password.length()] = "";
  email.toCharArray(email_char, email.length());
  password.toCharArray(password_char, password.length());
  Serial.println(email_char);
  Serial.println(password_char);
  auth.user.email = email_char;
  auth.user.password = password_char;

  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);
  
  file.close();
}
``` 
This error only occurs if I insert those 2 pieces of code.

mobizt, 2021-02-05:
You can't use BLE together with WiFi in Arduino platform unless you used in ESP-IDF with some some configuration for switching.

It uses the same radio and the flash used for BLE program and library is very huge and exceed to 114% space.

Search the google to extend the program space for BLE program.

You can only switch BLE on and WiFI off during get data from BLE and switch WiFi on and BLE of when connect to server.

You may need to reboot for BLE and WiFi switching.

closed 2021-02-05 by mobizt

Niltonsf, 2021-02-05:
@mobizt 
Hmm, but for example I have the data set already with SPIFFS, the bluetooth is not connected anymore, and I get this error still. Is it because I didn't create a connection with wifi or it is because of BLE?

mobizt, 2021-02-05:
You can't enable BLE and WiFi at the same time.

mobizt, 2021-02-05:
The BLE program when you include the BLE library exceed the program space in flash and may caused the error of user data flash access.

mobizt, 2021-02-05:
You should try the basic making https client to connect to some server and BLE integration test before.