mobizt/Firebase-ESP32 #146: ENHANCEMENT

petkomatus, 2021-02-18:
Can you help me send a message to Firebase.setString (fbdo, "message", "\" Arduino \ nESP32 \"");

i don't know about the top ride

Well thank you

mobizt, 2021-02-18:
I don't see any problem.
If you want to construct string from hard code the character inside quotes e.g. "your string" and your string contains quote " you should make a escape string by place \ before " e.g.

String str = "abcde \\"fghi\\" jklmn"; 

It will interpret as abcde "fghi" jklmn.

closed 2021-02-18 by mobizt

petkomatus, 2021-02-19:
i need to send to firebase in this shape
[image]

mobizt, 2021-02-19:
You should search Google for escape sequence string.

Test it by print to serial before sending.

String s1 = "\\"Arduino\nESP32\\"";

Serial.println(s1) 

This will pint  
**\"Arduino
ESP32\"**

 String s2 = "\\\\\\"Arduino\\nESP32\\\\\\""; 

This will pint  
**\\"Arduino\\nESP32\\\"**

petkomatus, 2021-02-19:
well thank you