mobizt/Firebase-ESP32 #148: Internet connection problem

am-canes, 2021-02-24:
I'm having a problem when ESP32 is connected to the wifi but it doesn't have internet connection.
In this case, the following message is showing in serial monitor:

`[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for xxxxxxxx.firebaseio.com
[E][FB_WCS32.cpp:160] connect(): start_ssl_client: -1`

I am using WiFi.mode (WIFI_AP_STA) because I am also using ESP32 as an access point to configure wifi credentials and another parameters.
But when I have this internet connection problem, the code is blocked on firebase function and my local page doesn't work anymore. And because of that I can't change wifi credentials. 

Is there any way in the library that I can identify the problem with the connection and stop the process to not block my local page and allow the user to change the wifi?

Thank you.

mobizt, 2021-02-24:
Nothing actually block as you understand in ESP32 due to it can assign others task for multitasking and it has two cores cpu.  You can create the new RTOS task to do your work.
Firebase call has the highest priority of the library when it trying to send request and response. 

Library does not has any blocking code unless the internal operation of SSL client. The blocking operations in SSL client (mbedTLS) happen duringg SSL handshaking, reading and writing data which all are blocking (waiting for response) process in secured connection. 

You can call fbdo.pauseFirebase(true) to by pass the Firebase call anytime you want. And call pauseFirebase(false) later for enanble the Firebase call.

closed 2021-02-24 by mobizt

am-canes, 2021-02-25:
Yes, I'm already using the other core to control critical sensors. 
So I need all the wifi, local webserver and firebase tasks to work in the other core.

I will try to identify this situation with internet connection problem and use pauseFirebase.

Thank you.

mobizt, 2021-02-25:
I will not block unless you use the Firebase in main task calling from set up and loop.

am-canes, 2021-02-25:
I'm calling everything on loop, because wifi may not be connected or configured on the start and can be done later. Therefore, I can't garantee that wifi is connected on setup.
So, when wifi is connected I initialize firebase and stream, only once:  

`Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
 Firebase.reconnectWiFi(true);
 path = "/" + devMacId + "/config";
   
 initData();

  if (!Firebase.beginStream(firebaseData1, path))
  {   
      initStream = false;
      Serial.println("------------------------------------");
      Serial.println("Can't begin stream connection...");
      Serial.println("REASON: " + firebaseData1.errorReason());
      Serial.println("------------------------------------");
      Serial.println();
  }
  else
    initStream = true;

   Firebase.setStreamCallback(firebaseData1, streamCallback, streamTimeoutCallback, 10000);  `

On initData() I'm seting some values with setJSON.

Then, when I have changes I use updateNode to update.

Do you thing because of that if I use pauseFirebase can I have a problem? Or is it okay?

mobizt, 2021-02-25:
Since you set the stream callback, the stream task runs in the RTOS task in core 1 which will be ok and not blocking main task.

Don't use firebaseData1 for other Firebase calls because it used in stream task already and not need to pause it.

If you use other Firebase Data object for Firebase calls (read/store operation) and use it in main task, you can pause it to bypass the operation.

am-canes, 2021-03-01:
Yes, I'm using another FirebasaData object to upload data. 

But I observed that my problem is occuring with the stream too. 

Now, I updated platform-espressif32 to latest version 3.1.0 that updated Arduino core to v1.0.5 and the error is occuring when wifi is disconnected too.
The web configuration portal doesn't work when wifi disconnects and the stream was already working. The same occurs if wifi connects but has internet connection problem.

I tried using endStream when wifi disconnects but it didn't work.

Maybe it is something on the core side. Do you have any suggestion that can solve this?

mobizt, 2021-03-01:
From the first post, you mention about blocking code and I provide the solution to bypass all Firebase calls and stream operation too. 

You should debug your Webserver code instead because when you pause the Firebase Data object, all functions will exit immediately, no code execution.

I don't have no suggestion because it's not related to the library, unless check your program flow and memory concerns.

am-canes, 2021-03-05:
Ok, about the problem when wifi disconnects I could solve, it was a conflict between my code and reconnectWiFi.

But the other problem I couldn't solve even with pauseFirebase. Because it looks like that the code is blocked on something that is calling start_ssl_client all the time. Then it can't reach my code to pause.
Sometimes it could reach the code, execute pauseFirebase and keeps calling start_ssl_client. 

If I understood correctly, the stream is trying to connect and calls start_ssl_client right? There is a way to stop it, or to know which function is calling start_ssl_client ?

mobizt, 2021-03-05:
> You can call fbdo.pauseFirebase(true) to by pass the Firebase call anytime you want. And call pauseFirebase(false) later for enanble the Firebase call.

Did you pause the stream's Firebase Data object?

mobizt, 2021-03-05:
The SSL client created inside the object, you can control it by set the reference and pointer to it too.

mobizt, 2021-03-05:
This should be the last discussion about this.