mobizt/Firebase-ESP32 #149: Corrupt Heap

giorgio80, 2021-02-24:
**Describe the bug**
getting corrupt heap on document writing 
**To Reproduce**

unfortunately I cannot reproduce it as it seems to be random but I espExceptiondecoder the output

**Expected behavior**
during running an update esp32 simply hangs and restart 

**Screenshots**
If applicable, add screenshots to help explain your problem.

[https://i.postimg.cc/nhNK2hbC/Screenshot-2021-02-24-at-11-46-03.png](url)

**IDE and its version:**
 Using platform with latest espressif version of idf (2.0)

mobizt, 2021-02-24:
You should provide the code and check your free memory before calling Firebase functions.
No thing can help based on your info.

closed 2021-02-24 by mobizt

giorgio80, 2021-02-27:
is what I 'm doing, 

btw indicating seems related to function: 

void FirebaseESP32::delS(char *p)
{
  if (p != nullptr)
    delete[] p;
}

it seems to hangs when launch the delete[] p

can I ask you what's the role of this function to start debugging ?

mobizt, 2021-02-28:
I don't understand what you are trying to do.

Please learn the basic C or C++ programming.

mobizt, 2021-02-28:
The error from decoded trace is about dynamic memory allocation by calling **new** to allocate the memory then the **delete** was called to free the memory.

The error may occur when not enough memory for allocation with the desired amount due to you misused the stream.

The data at that node in the database which  you're trying to stream is too large to handle by the limited memory device.

You should reduce the data to get from stream.

mobizt, 2021-02-28:
You should also update the ESP32 Core SDK.