mobizt/Firebase-ESP32 #150: BUG

ICTAddict, 2021-02-26:
**Describe the bug**
When using Email_Password from the examples, setting the realtime database rules fails and shows the following on serial monitor: 

Read database ruless...
Set database rules...
Failed to edit the database rules, bad request
------------------------------------

**Expected behavior**
database rules should be changed to grant the user read and write access to "/UsersData2/xxxx" where xxxx is uid of the user

**IDE and its version:**
 - Arduino
 - Version 1.8.13

**ESP32 Arduino Core SDK version**
 - 1.0.4

mobizt, 2021-02-26:
Database rules access requires admin rights scope authentication i.e. OAuth2.0 token and legacy token (database secret).

mobizt, 2021-02-26:
Why you expect user to access the database rules?
It's nonsense and not secured to allow that. That's why Google design the security rules for all Firebase products to control the user access.

Please read the the Firebase doc for security rules.

closed 2021-02-26 by mobizt

mobizt, 2021-02-26:
When you open the issue, if you not sure about the problem, please choose Help instead of bugs.