mobizt/Firebase-ESP32 #151: HELP: Infinite connection loop

Grabt234, 2021-03-08:
I am currently using your code to try connect to firebase. The code works most of the time but sometimes enters into an infinite loop while waiting for "ready". Do you have any idea why firebase would not be able to connect, why it would enter an infinite loop and how one could fix it? 

`void GatewayLayer4::firebaseInit()
{
/* Assign the certificate data (optional) */
    //config.cert.data = rootCACert;

    /* Assign the project host and api key (required) */
    config.host = FIREBASE_HOST;
    config.api_key = API_KEY;

    /* Assign the sevice account credentials and private key (required) */
    config.service_account.data.client_email = FIREBASE_CLIENT_EMAIL;
    config.service_account.data.project_id = FIREBASE_PROJECT_ID;
    config.service_account.data.private_key = PRIVATE_KEY;

    uint8_t selfNameArr[SENSOR_LENGTH] = {0};
    uint8_t *ptr_selfNameArr = &selfNameArr[0];
    selfName(ptr_selfNameArr);
    std::string selfNameString = (char*)selfNameArr;

    auth.token.uid = selfNameString;

    /** Assign the custom claims (optional)
     * This uid will be compare to the auth.token.premium_account variable
     * (for this case) in the database rules.
    */
    auth.token.claims.add("premium_account", true);
    auth.token.claims.add("admin", true);

    Firebase.reconnectWiFi(true);
    Firebase.setReadTimeout(fbdo, 1000 * 120);
    fbdo.setResponseSize(8192);

    String base_path = //removed

    Firebase.begin(&config, &auth);

    path = base_path + auth.token.uid.c_str();

    struct token_info_t info = Firebase.authTokenInfo();
    String ready = "ready";
    while (ready.compareTo(getTokenStatus(info)))
    {
        delay(1000);
        #ifdef LL4_DEBUG_FIREBASE
        Serial.println("GatewayLayer4::firebaseSetup(): Connecting to Firebase");
        #endif
        info = Firebase.authTokenInfo();
    }
    
}`

mobizt, 2021-03-09:
I will update soon and let you know.

mobizt, 2021-03-13:
The library update, v 3.8.14 is now available. 

The infinite loop issue was fixed.

The callback function can be assigned for token generation as [this](https://github.com/mobizt/Firebase-ESP-Client/blob/f0c05263719c54a592083c0474eedd4f96efe6f9/examples/Authentications/Access_Token_from_File/Access_Token_from_File.ino#L100-L103) and [this](https://github.com/mobizt/Firebase-ESP-Client/blob/f0c05263719c54a592083c0474eedd4f96efe6f9/examples/Authentications/Access_Token_from_File/Access_Token_from_File.ino#L158-L178).

Please update.

closed 2021-03-13 by mobizt