mobizt/Firebase-ESP32 #154: BUG Cloud_Messaging.ino Example

Niassamond, 2021-03-28:
I just updated the new version of the library, and my cloud messaging functions stopped working. Im currently using 3.8.20 Firebase ESP32 Client version.
When i updated, i started fixing my old code(months old, dont know what library was used back then) i checked the examples and was able to fix everything in my code, except for the FCM.
I checked the example, and programed my ESP32 with it, using my ssid, password, host, secret, key and token. other than that nothing was altered, and it has the same issue that my code: I get the notification in my app, but the notification is blank. No title, no body nothing, just a floating ballon. I also noted that the notification only comes when the app is opened, when its closed nothing arrives. I think the second problem that im getting no notifications when the app is closed might be due to the fact that the notification is completly empty. can someone help me with this issue?
Thanks in advance!

mobizt, 2021-03-29:
Thanks for report this bug.

I will fix it and update soon.

mobizt, 2021-03-29:
The issue fixed and the library was updated to v3.8.21.

Please update.

closed 2021-03-29 by mobizt