mobizt/Firebase-ESP32 #155: Firebase Cloud Messaging use channel_ID

Niassamond, 2021-03-29:
Hello, i'd like some help implementing a channel_id to my notifications on firebase library, im trying to use the 
firebaseData1.fcm.setDataMessage("{\"channel_id\":\"channel1\"}"); command, but it doesnt set the right channel id.
I also tried   fbdol.fcm.addCustomNotifyMessage("channel_id", "channel1"); but this command gives me errors.
i already did it work with setDataMessage before but i think im doing something wrong now.

mobizt, 2021-03-29:
Please update the library to v3.8.22.
The earlier version has a fcm bugs.

You can use addCustomNotifyMessage as  you want.

Niassamond, 2021-03-29:
Im currently using the V3.8.22
When i use the command 
fbdol.fcm.addCustomNotifyMessage("channel_id","channel1"); 
when i do this i get the following error:
exit status 1
'fbdol' was not declared in this scope

mobizt, 2021-03-29:
You are typo.
Check what's the name it declared, fbdo1 or fbdol.
It should be fbdo1.

closed 2021-03-29 by mobizt

mobizt, 2021-03-29:
Check your code what's the name of FirebaseData object [here](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Cloud_Messaging/Cloud_Messaging.ino#L31).

Are you never used Firebase RTDB before or you using the library only for FCM?

Niassamond, 2021-03-29:
Im using the RTDB , and you're right i did messed up the l with a 1, now its fixed and no longer get errors from the Arduino IDE, but my notifications are still not on channel1 from channel id, i think i might be doing something wrong.... before i updated your library i was doing the channel id like this:
firebaseData1.fcm.setDataMessage("{\"channel_id\":\"channel1\" \"title\":\"Interfone\" }");
now when i try to use it, i get the ESP32 error :
FAILED
13:19:40.638 -> REASON: bad request
When i added the correct line 
fbdo1.fcm.addCustomNotifyMessage("channel_id", "channel1"); 
both arduino and ESP32 Work, but my notification is still on default channel_id instead of channel1.

Niassamond, 2021-03-29:
i honestly think im doing something wrong when defining the channel_id, im a bit of a noob when it comes to arduino... and i really appreciate the time you're taking to help me.
right now the code is:
fbdo1.fcm.addCustomNotifyMessage("channel_id", "channel1");
The arduino is ok with it, and the ESP sends the notification without any problems.
But the notification is on default channel instead of channel1

mobizt, 2021-03-29:
You need to check the message.

The custom notify message may not allow.

When I test, the custom message when received it goes under the notification data instead with the key like gcm.notification.channel_id.

You should check the [FCM document](https://firebase.google.com/docs/cloud-messaging/http-server-ref#notification-payload-support) for supported payload.

It support only "android_channel_id" .

Niassamond, 2021-03-29:
hmmm, im going to test it right now! thanks for the reply

Niassamond, 2021-03-29:
It worked! Thanks a lot!!!!!!!