mobizt/Firebase-ESP32 #157: HELP Can't setBool while MultiPathStream is set up

nathansbird, 2021-04-03:
I'm using this library to listen for data to change and then immediately alter that data. I'm using the multipath stream feature to listen to a path with three children. Everything seems to be fine on the reading end but when I try to write, I get the error `http connection was used by other processes`

I don't have any issues writing prior to listening. Is there a way to run multiple connections or something? That doesn't make sense that a single stream would block the writing. You even have broken example projects that make it seem possible.

mobizt, 2021-04-03:
Please don't blame anything base on your understandin.

Please post your code here.

You must not share the Firebase Data object used by stream (multiPathStream) with other Firebase calls.

mobizt, 2021-04-03:
Please respect the hard work and stop blaming.

closed 2021-04-03 by mobizt

mobizt, 2021-04-03:
See [this example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/MultiPath_stream/MultiPath_stream.ino) how to use multi path stream correctly.

nathansbird, 2021-04-03:
@mobizt I'm sorry, I didn't mean to come across that way, my bad! I do respect your hard work, this library is amazing considering the intense limitations of the ESP32! 👏🏻

mobizt, 2021-04-03:
I get it now and sorry for misunderstanding.

You should use the different Firebase Data object for other Firebase calls.

nathansbird, 2021-04-03:
Can you elaborate on what you mean by not passing the data object? Here's what I'm doing in simplified code:

```
//Inside a function, after getting the auth token
Firebase.RTDB.beginMultiPathStream(&fbdo, path.c_str(), childPath, 5);
Firebase.RTDB.setMultiPathStreamCallback(&fbdo, streamCallback, connectLost);

void streamCallback(MultiPathStreamData stream){
  size_t numChild = sizeof(childPath)/sizeof(childPath[0]);
  for(size_t i = 0; i < numChild; i++){
    if (stream.get(childPath[i])){
      if (stream.dataPath == "/ping" && stream.value == "false"){
        Firebase.RTDB.setBool(&out, ("/devices/68f244/ping").c_str(), true);
      }
    }
  }
}
```

A few things to note:
I've placed Serial prints inside the nested if and they print, the logic is correct
I've moved the setBool line to the beginning and it works
As noted at the top, the error from `out` is `http connection was used by other processes`

mobizt, 2021-04-03:
The **fbdo** object which you pass to **beginMultiPathStream**.
The fbdo should not use in other Firebase calls e.g. get, set, push etc.

You should define the other Firebase Data object e.g. fbdo2, for other Firebase calls.

nathansbird, 2021-04-03:
Okay I see. Yeah, I'm not doing that. Any other ideas?

mobizt, 2021-04-03:
In this case the callback function is called when the internal blocking status flag that states the completion of payload handling process  is not clear before calling other Firebase calls.  This blocking flag will reset later after payload handling process was completed.

I will fix this in the next library update.

Based on your use case, you can edit the library at [this line](https://github.com/mobizt/Firebase-ESP32/blob/master/src/rtdb/FB_RTDB.cpp#L4274).

from 

```cpp
void FB_RTDB::sendCB(FirebaseData *fbdo)
{
    if (fbdo->_dataAvailableCallback)
    {

```
to

```cpp
void FB_RTDB::sendCB(FirebaseData *fbdo)
{
    Signer.getCfg()->_int.fb_processing = false; //allow other subsequence requests before calling the callback.

    if (fbdo->_dataAvailableCallback)
    {
```

mobizt, 2021-04-03:
The library update v3.8.25 is available with this issue fixed.

Please update.

nathansbird, 2021-04-03:
Yep, that did it. Thank you!