mobizt/Firebase-ESP32 #159: HELP! LOTS OF ERRORS AFTER UPDATING

zeeshan0309, 2021-04-14:
Please Help! I updated Firebase-ESP32 and FirebaseJson libraries today by arduino library manager, and all of a sudden a VERY long list of errors is popping up and code is not compiling. Before today's update i was probably using the versions which were released around Dec '20/Jan '21. Error message:

`libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_finalize()':
c:\users\zeeshan\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h:923: multiple definition of `FirebaseJson::_finalize()'
libraries\FirebaseJson\FirebaseJson.cpp.o:C:\Users\zeeshan\Documents\Arduino\libraries\FirebaseJson\src/FirebaseJson.cpp:105: first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_init()':
FirebaseJson.cpp:(.text._ZN12FirebaseJson5_initEv+0x0): multiple definition of `FirebaseJson::_init()'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson5_initEv+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::setJsonData(String const&)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson11setJsonDataERK6String+0x0): multiple definition of `FirebaseJson::setJsonData(String const&)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson11setJsonDataERK6String+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_setJsonData(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson12_setJsonDataERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): multiple definition of `FirebaseJson::_setJsonData(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson12_setJsonDataERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::setBufferLimit(unsigned int)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson14setBufferLimitEj+0x0): multiple definition of `FirebaseJson::setBufferLimit(unsigned int)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson14setBufferLimitEj+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_toStdString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, bool)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson12_toStdStringERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb+0x0): multiple definition of `FirebaseJson::_toStdString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, bool)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson12_toStdStringERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::int_toStdString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, bool)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson15int_toStdStringERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb+0x0): multiple definition of `FirebaseJson::int_toStdString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, bool)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson15int_toStdStringERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_add(char const*, char const*, unsigned int, unsigned int, bool, bool)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson4_addEPKcS1_jjbb+0x0): multiple definition of `FirebaseJson::_add(char const*, char const*, unsigned int, unsigned int, bool, bool)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson4_addEPKcS1_jjbb+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_addString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson10_addStringERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_+0x0): multiple definition of `FirebaseJson::_addString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson10_addStringERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES7_+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::add(String const&, String const&)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6StringS2_+0x0): multiple definition of `FirebaseJson::add(String const&, String const&)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6StringS2_+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::add(String const&, char const*)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6StringPKc+0x0): multiple definition of `FirebaseJson::add(String const&, char const*)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6StringPKc+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_addInt(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson7_addIntERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEi+0x0): multiple definition of `FirebaseJson::_addInt(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson7_addIntERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEi+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::add(String const&, int)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6Stringi+0x0): multiple definition of `FirebaseJson::add(String const&, int)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6Stringi+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::add(String const&, unsigned short)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6Stringt+0x0): multiple definition of `FirebaseJson::add(String const&, unsigned short)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6Stringt+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_addFloat(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson9_addFloatERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEf+0x0): multiple definition of `FirebaseJson::_addFloat(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson9_addFloatERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEf+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::add(String const&, float)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6Stringf+0x0): multiple definition of `FirebaseJson::add(String const&, float)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN12FirebaseJson3addERK6Stringf+0x0): first defined here
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJson::_addDouble(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double)':
FirebaseJson.cpp:(.text._ZN12FirebaseJson10_addDoubleERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEd+0x0): multiple definition of `FirebaseJson::_addDouble(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const& [... cut ...]
libraries\Firebase_ESP32_Client\json\FirebaseJson.cpp.o: In function `FirebaseJsonArray::set(String const&, FirebaseJsonArray&)':
FirebaseJson.cpp:(.text._ZN17FirebaseJsonArray3setERK6StringRS_+0x0): multiple definition of `FirebaseJsonArray::set(String const&, FirebaseJsonArray&)'
libraries\FirebaseJson\FirebaseJson.cpp.o:FirebaseJson.cpp:(.text._ZN17FirebaseJsonArray3setERK6StringRS_+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.`

mobizt, 2021-04-14:
Please uninstall FirebaseJson library because it's already included in the Firebase library.

mobizt, 2021-04-14:
Try these fixes

- Remove the include file FirebaseJson.h from FirebaseESP32 sketch.

- If you want to use FirebaseJson without Firebase ESP32 Client library, copy two files in **C:\Users\USER_NAME\Documents\Arduino\libraries\FirebaseJson\src**
to **C:\Users\USER_NAME\Documents\Arduino\libraries\Firebase_ESP32_Client\src\json** by replaced the existing files.

This will be fixed in the next library update.

zeeshan0309, 2021-04-14:
> Please uninstall FirebaseJson library because it's already included in the Firebase library.

So I did this and removed it's reference from my main code, and its working just fine now. Thanks a whole lot!

mobizt, 2021-05-01:
Library updated with the issues fixed.

closed 2021-05-01 by mobizt