mobizt/Firebase-ESP32 #160: BUG

rockmanexze, 2021-04-20:
After the latest update i ran into this error 

In file included from C:\Users\ad\Documents\Arduino\ArduinoFirebase-2\ArduinoFirebase-2.ino:4:0:

C:\Users\ad\Documents\Arduino\libraries\Firebase_ESP32_Client\src/FirebaseESP32.h:1691:45: error: 'fb_esp_fcm_msg_type' has not been declared

   bool handleFCMRequest(FirebaseData &fbdo, fb_esp_fcm_msg_type messageType);
                                                                             ^
Is there a fix to this? or should i just roll back the updates

mobizt, 2021-04-21:
This is not a bug unless the installation problem.

I check and test it already and the enum is already here.
https://github.com/mobizt/Firebase-ESP32/blob/4ee7acd7b8b62e719207d139b87a8dbb54377945/src/session/FB_Session.h#L44

Uninstall and install again. If the error still existed, try [this example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Cloud_Messaging/Cloud_Messaging.ino).

If no error with the example and error only with your working code you should check the third party libraries included.

You should copy the full compile error and post here.

mobizt, 2021-04-21:
Even with the v3.8.25 the enum is already here as the same with latest version.
https://github.com/mobizt/Firebase-ESP32/blob/005f47409de46f4900b49662a9729d3c3690a437/src/session/FB_Session.h#L44

closed 2021-04-21 by mobizt